AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #618: Stop Charging When Sleeping not working

Choongyangl, 2022-03-08:
I am a 14' Macbook Pro user. Recently just purchased the Al Dente Pro however the Stop Charging When Sleeping not working.

My charge limit is 65%, i plugged in the charger when 60% and put my laptop to sleep.

Then I wake it on after 5 minutes, the laptop eventually charge to 63% which mean the stop charging when sleeping is not working.

MatthiasKerbl, 2022-03-08:
Hi @Choongyangl,

Which version of AlDente Pro are you using?
Do you have the feature "Disable Sleep until Charge Limit" enabled?

Cheers,
Matthias

MatthiasKerbl, 2022-03-08:
Hi @Choongyangl,

Thank you for providing more information.

Please follow these steps:
1. Do a complete uninstallation of AlDente Pro: https://apphousekitchen.com/how-to-uninstall-aldente-pro/
2. Reinstall AlDente Pro: https://github.com/davidwernhart/AlDente/releases
3. After reinstalling, use the same Charge Settings you are using before and set the charge limit at least 20% above the current battery percentage.
4. Close the lid and wait for at least 30 minutes.
5. Open the lid again and check if the current battery percentage is within the range (+-5%) of the battery percentage your MacBook had when you closed the lid.

Let me know what happened.

Cheers,
Matthias

Choongyangl, 2022-03-09:
Hi @MatthiasKerbl ,

I have just tried to uninstall it.

But it is not working still.

Choongyangl, 2022-03-09:
Hi @MatthiasKerbl 

After reinstall the software, sailing mode is not working as well.

MatthiasKerbl, 2022-03-09:
> Hi @MatthiasKerbl ,
> 
> I have just tried to uninstall it.
> 
> But it is not working still.

Hi @Choongyangl,

Please be more specific. What exactly happened?

Choongyangl, 2022-03-09:
Hi @MatthiasKerbl ,

Below is my al dente pro setting.
Charge Limit: 65%
Sailing Mode Enabled with Different 7%
Stop charging when sleeping Enabled

For Sleep while charging, its not working. My battery was 55% and plugged in the charger and put my computer on sleep for half hour, it charge mine laptop to 65%.

For Sailing Mode, once my battery charged to 65% it never sail to 58%. it keep stucked at 65% for the whole day as long as the charger is connect to my macbook.

All these issues happened even after I reinstall the application

MatthiasKerbl, 2022-03-09:
Hi @Choongyangl,

It looks like there is something off with the "Stop charging when sleeping" feature. We are looking into this right now.

Regarding the Sailing Mode, this feature is not intended to discharge your battery to 58%. It will just not charge the battery when you unplug it for a bit or if the battery charge falls back on its own a little bit. Please read our article about the Sailing Mode: https://apphousekitchen.com/feature-explanation-sailing-mode/

Choongyangl, 2022-03-09:
Hi @MatthiasKerbl 

thanks, pls let me know if there is any update on the Sleep while charging issue

Choongyangl, 2022-03-09:
Hi @MatthiasKerbl 

thanks, pls let me know once the issue is fixed for stop charging when sleeping feature

MatthiasKerbl, 2022-03-09:
Hi @Choongyangl,

We have just released an update for AlDente Pro which should fix the issue you encountered.
https://github.com/davidwernhart/AlDente/releases/tag/1.15.2

Cheers,
Matthias

closed 2022-03-09 by MatthiasKerbl

Choongyangl, 2022-03-09:
Hi @MatthiasKerbl 

Issue fixed.

Thanks.

pbond630, 2022-03-12:
I'm still having this issue, **even after update to 1.15.2**. Macbook Pro charges to 100%, ignoring the 70% limit set in Al Dente, if plugged in when Macbook is sleeping. 'Stop Charging When Sleeping' is enabled in settings.
I reinstalled the helper after updating app to 1.15.2, but issue persists. The issue started in approx the past month or so. I have a debug file I can send if that helps.

Macbook Pro Apple SIlicon 14", Monterey 12.2.1.

MatthiasKerbl, 2022-03-14:
> I'm still having this issue, **even after update to 1.15.2**. Macbook Pro charges to 100%, ignoring the 70% limit set in Al Dente, if plugged in when Macbook is sleeping. 'Stop Charging When [...]

Hi @pbond630,

Please send us the debug file when it occurs again.

pbond630, 2022-03-17:
Hi Matthias

This has happened again. I have a debug file saved. Could you let me know the address to send it to. Thx.

MatthiasKerbl, 2022-03-17:
> Hi Matthias
> 
> This has happened again. I have a debug file saved. Could you let me know the address to send it to. Thx.

Hi @pbond630,

Please send it to news@apphousekitchen.com