AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #620: [Feature Request] Show Current Battery Temperature in Al Dente Popup Window

mitchellisdumb, 2022-03-09:
**Is your feature request related to a problem? Please describe.**
It takes a few clicks to see the current battery temperature.

**Describe the solution you'd like**
I would like the status readout in the Al Dente popup window to include the current battery temperature.

**Describe alternatives you've considered**
I could install iStatMenus or similar.

**Additional context**

Stickyhavr, 2022-03-20:
I agree. Currently you have to dig down into Settings then Charge Settings just to see the current temp. It would be great if you could see it on the first click of the menu bar. Or, optionally, have it displayed in the menu bar along with the status and percentage, as requested in #489

basmets6, 2022-04-17:
+1 on this one

rusteze89, 2023-09-30:
+1 on this particularly now that there is the option to display Power Flow. The Stats menu could have the option of displaying Power Flow **and Temperature** on the popup window.

closed 2024-02-09 by MatthiasKerbl