AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #622: Al dente doesn't sleep after it's fully charged

Grolubao, 2022-03-10:
**Describe the bug**
I have Al Dente version 1.15.2 and if I close the lid of my MB Pro after it reaches the limit it never actually sleeps, it continues awake.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable all the settings on Charge Settings using Al Dente Pro
2. Close the lid.
3. After reaching the charge limit, it actually never goes to sleep afterwards

**Expected behavior**
After reaching the limit it should go to sleep

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model - MB Pro 15" Late 2016
 - macOS version 12.2.1
 - AlDente Pro or AlDente Free? Pro
 - AlDente Version? 1.15.2
 - List of all enabled/disabled settings in AlDente: All enabled

Grolubao, 2022-03-11:
> Hi @Grolubao <https://github.com/Grolubao>,
>
> I have just tested it on my 15" Late 2016 MacBook Pro and it worked
> without any issues.
> How long did you wait for your MacBook to go into sleep after it had
> reached the charge limit? Because it can take a while until it goes into
> sleep.
>
> Cheers,
> Matthias

Hi Matthias,

I left it during the night and when I woke up the fans were still spinning
and clearly it didn't go to sleep. It was charged to the set amount of 80%
which is good, but again it was not sleeping, I just expected that once it
reaches the limit it would then go to sleep but this was not the case.

Thanks

MatthiasKerbl, 2022-03-12:
> Hi Matthias, I left it during the night and when I woke up the fans were still spinning and clearly it didn't go to sleep. It was charged to the set amount of 80% which is good, but again it was not [...]

Hi @Grolubao,

Thank you for the details. Your MacBook should definitely go to sleep when the charge limit is reached. Could you generate and share a debug file the next time this happens? https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Cheers,
Matthias

nheeren, 2022-04-11:
propbably related to #640?

MatthiasKerbl, 2022-04-13:
Hi @Grolubao,

Do you still encounter this issue with our newest version AlDente 1.16?

Cheers,
Matthias

Grolubao, 2022-05-13:
Hi David,

Today the same behaviour happened. So my MB Pro 15" didn't actually sleep
at all even though it already reached the desired charge. When I woke up it
was super hot.

Herewith the debug file

Thanks

MatthiasKerbl, 2022-09-02:
> Hi David, Today the same behaviour happened. So my MB Pro 15" didn't actually sleep at all even though it already reached the desired charge. When I woke up it was super hot. Herewith the debug file Thanks On Sat, Mar 12, 2022 at 9:07 AM Matthias Kerbl ***@***.***> wrote:
> […](#)

Hi @Grolubao,

We have made many new improvements with our newest version AlDente 1.18.
Do you still experience this issue with this version?

Cheers,
Matthias

Grolubao, 2022-09-12:
Hi there,

Just tested and indeed the behaviour is the same. When I woke up this
morning the laptop was extremely hot. The defined charging percentage was
correct which is good, but clearly the laptop did not sleep at all

Best

Grolubao, 2022-09-20:
Herewith the debug file:

Grolubao, 2022-09-20:
Weird, via email it does not attach. Herewith it
[aldentepro-debug-2022-09-20-07_38_09.pasta.zip](https://github.com/davidwernhart/AlDente/files/9607592/aldentepro-debug-2022-09-20-07_38_09.pasta.zip)

MatthiasKerbl, 2022-09-29:
Hi @Grolubao,

We have looked through your debug file and could unfortunately not find the issue there.
Could you check in macOS Settings/Battery and PowerAdapter what you chose for "Turn display off after:"?

Cheers,
Matthias

Grolubao, 2022-09-30:
Hi @MatthiasKerbl,
That is very strange, but again the settings you mention should not affect it since it's settings whilst the laptop is awake not "sleeping". In my other laptop I have the same settings and it works just fine

[image]

MatthiasKerbl, 2022-09-30:
Hi @Grolubao,

I am very much sure that this is the reason why your MacBook does not go to sleep when it is fully charged.
Please set the "Turn display off after:" setting to something like 15 minutes and do the same for the Power Adapter settings and try again if your MacBook goes to sleep when it is fully charged this time.

Cheers,
Matthias

Grolubao, 2022-11-03:
Hi @MatthiasKerbl, what I really don't get is that AlDente should essentially override the system settings to make sure it goes to sleep no matter what. If you look at AlDente settings you assume this:

[image]

It disables sleep until the Charge Limit, so no matter the "Turn display off after" it should still disable sleep, however when the limit is reached then it should sleep and it isn't

[image]

What is the problem here?

MatthiasKerbl, 2022-11-03:
Hi @Grolubao,

> what I really don't get is that AlDente should essentially override the system settings to make sure it goes to sleep no matter what.
I disagree with this statement. In general, we are trying to take control of macOS as little as possible but as much as necessary to reach the goal of AlDente and its features.
This means in regards to the feature Disable Sleep until Charge Limit that we take control over the sleeping behavior until the charge limit is reached. When the charge limit is reached, we give back control over the sleeping behavior to macOS. Therefore, what happens to sleep after the charge limit is reached depends on your sleep settings in macOS.

Now coming back to your issue that your MacBook does not sleep when the charge limit is reached. I was wrong in the assumption that the cause of this behavior is because you have the "Turn display off after:" setting set to Never. After extensive testing, it seems that the real cause is the setting "Prevent your Mac from automatically sleeping when the display is off" which you have enabled. Therefore, my suggestion, if you want your MacBook to go to sleep when the charge limit is reached, you have to disable this setting.

Cheers,
Matthias

Grolubao, 2022-11-04:
Hi @MatthiasKerbl I have just disabled it before going to be and when I woke up the laptop was extremely hot and clearly didn't shut down the whole night. Please find attached the debug log

MatthiasKerbl, 2022-11-04:
> Hi @MatthiasKerbl I have just disabled it before going to be and when I woke up the laptop was extremely hot and clearly didn't shut down the whole night. Please find attached the debug log

Hi @Grolubao,

Just to be sure, you disabled "Prevent your Mac from automatically sleeping when the display is off" and your MacBook still did not sleep when the Charge Limit was reached?
Please attach the debug file here in GitHub again.

Cheers,
Matthias

Grolubao, 2022-11-04:
[aldentepro-debug-2022-11-04-07:25:20.pasta.zip](https://github.com/davidwernhart/AlDente/files/9937120/aldentepro-debug-2022-11-04-07.25.20.pasta.zip)

Grolubao, 2022-11-09:
This is also happening on my other laptop with AlDente which is an Intel one. Here's the debug file there. I'll have to stop using AlDente at least temporarily until this is fixed otherwise the laptop is never going to sleep at all 😞 
[aldentepro-debug-2022-11-09-08:18:21.pasta.zip](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/9968895/aldentepro-debug-2022-11-09-08.18.21.pasta.zip)

MatthiasKerbl, 2022-11-09:
Hi @Grolubao,

We might have found a little bug with the help of your debug files that could potentially have something to do with the issue you reported. We will release the fix with the next update.
In the meantime, however, we are still not able at all to reproduce the bug you encounter on any of our Intel or Apple Silicon MacBooks. Could you explain to me in as much detail as possible what you did to encounter this issue and send me screenshots of your AlDente settings and macOS battery, power adapter, and sleep settings?

Thanks in advance,
Matthias

Grolubao, 2022-11-10:
Great, thanks for letting me know. Here are the settings for my Intel machine. Again this was happening on the Apple silicon one too
[image]
[image]
[image]

MatthiasKerbl, 2022-11-14:
Hi @Grolubao,

I have tested this scenario with exactly your settings thoroughly over the last couple of days and I am still not able to reproduce your issue. My MacBook always goes to sleep after a while and since you set the "Turn display off after:" setting to 3h it usually takes 3-4 hours after it is fully charged to go to sleep.
A couple of suggestions:

- I have macOS Ventura installed. If it is not an issue for you, I would recommend installing Ventura as well and trying to replicate this issue again.
- Is it possible that you have any other app installed that influences or might interfere with AlDente or macOS?
- Could you set the "Turn display off after:" setting to 10 min and test it again?

Cheers,
Matthias

Grolubao, 2022-11-30:
Hello @MatthiasKerbl, I really don't understand the behaviour, why sometimes it does sleep but others do not. This night happened the same, it just didn't go to sleep at all. Herewith the debug file

[aldentepro-debug-2022-11-30-04:51:54.pasta.zip](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/10119569/aldentepro-debug-2022-11-30-04.51.54.pasta.zip)

MatthiasKerbl, 2022-12-24:
Hi @Grolubao,

Did you try my suggestions above?

> I have macOS Ventura installed. If it is not an issue for you, I would recommend installing Ventura as well and trying to replicate this issue again.
> Is it possible that you have any other app [...]

Cheers,
Matthias

Grolubao, 2022-12-25:
Hi Matthias,

Unfortunately my Macbook Pro 15" is from 2016 so I can't install Ventura.
Is there a way I can pinpoint which app is not allowing it to sleep
properly? I'd expect that the debug file would reveal such app

I did Turn the display off after setting and it's still the same issue some
times

Thanks

MatthiasKerbl, 2023-03-12:
Hi @Grolubao,

Any updates on your issue?
Are you still experiencing this issue on our newest version AlDente 1.21.1?
If you are still experiencing this issue, please consider installing Sleep Aid(https://ohanaware.com/sleepaid/) to figure out which app is preventing your MacBook from sleeping. It has a 14-day free trial.

Cheers,
Matthias

Grolubao, 2023-03-13:
Hi Matthias,

First of all, thanks for checking in, I appreciate it.

I haven't experienced this behaviour again, but if I do I'll let you know
and I'll install Sleep Aid.

You can close the issue for now.

Cheers,
Raf