AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #624: MacBook shut down plugged in

slvrscoobie, 2022-03-12:
Been using Al dente for a while. Recently I upgraded to the latest version. Today the MacBook just shut off. It was sitting at 84% battery until then.  Power button resulted in red battery and plug in symbol. I plugged in it booted right back up saying it was at 85%. Checking Istats menu it shows it went to 2% from 84%. 
since I’m on the beta I needed an update anyway so updated and rebooted. Now it shows 99% but I’m trying to figure out why it shut off to begin with. Plugged in for weeks, months really sitting at 70%.

MatthiasKerbl, 2022-03-12:
Hi @slvrscoobie,

Looks like your battery calibration is off because you did not do any full cycles for a long time.

Here is what you can do from our readme:

HELP MY BATTERY CAPACITY DROPPED AND MY MACBOOK TURNS OFF WITH 50% LEFT
Keeping your battery at a lower percentage, such as 20-70%, over weeks without doing full cycles (100%-0%-100%) can result in a disturbed battery calibration. When this happens, your Macbook might turn off with 40-50% left or your battery capacity will drop significantly. However, this is only due to a disturbed battery calibration and not because of a faulty or degraded battery. To avoid this issue, we recommend doing at least one full cycle (0%-100%) every two weeks. Even if your battery calibration gets disturbed, doing 4-5 full cycles will recalibrate your battery and the capacity will go up again. AlDente Pro has a feature called Calibration Mode(https://apphousekitchen.com/feature-explanation-calibration-mode-2/), which will automatically do a full cycle when started. You can read more about the calibration of "smart" batteries here: https://batteryuniversity.com/article/bu-603-how-to-calibrate-a-smart-battery

Cheers,
Matthias

slvrscoobie, 2022-03-12:
maybe ill just go back to the old version, never had an issue with it on that version and id had for months. shutting down while plugged in seems like a pretty serious bug.

MatthiasKerbl, 2022-04-04:
Hi @slvrscoobie,

Could you resolve the issue with our instructions?

Cheers,
Matthias

slvrscoobie, 2022-04-04:
yes, however,  the first 2 cycles didnt even drain below 88% and got me worried. I followed the SMC reset again while after deleting the intel helper and the app, and then fully charged. that 3rd cycle it sat on 100% for like 3 hours then dropped to about 80% before shutting off, the 4th cycle it finally went below 80%. a 5th cycle was needed to finally get it down to ~20% without shutting off. now it seems back to normal having removed the app. having used Al Dente for years on the older version without any issues keeping it plugged in for long periods (months) ive removed it from my laptop and glad I never installed it on my M1 devices.

MatthiasKerbl, 2022-04-19:
Hi @slvrscoobie,

Again, this issue has nothing to do with different versions of AlDente or different MacBook models. It is just what happens when you leave your battery at a certain percentage for months without doing full cycles (100%-0%-100%). You can read more about this topic here: 

https://apphousekitchen.com/feature-explanation-calibration-mode-2/
https://batteryuniversity.com/article/bu-603-how-to-calibrate-a-smart-battery

This issue does not occur when you do at least one full cycle every two weeks or so. 

Cheers,
Matthias

closed 2022-04-19 by MatthiasKerbl