AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #628: Al Dente stopped working

ozmo1925, 2022-03-14:
Hi, if you have a question about AlDente, please use the Discussions section of Github: https://github.com/davidwernhart/AlDente/discussions
 Al Dente pro was working good but it doesn’t work anymore my battery is always at %100 percent I tried to restart Mac but nothing changed I am using Big Sur 11.6 any help?

MatthiasKerbl, 2022-03-14:
Hi @ozmo1925,

Please answer the following questions:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

radovansurlak, 2022-03-25:
@MatthiasKerbl 
**Describe the bug**

Given:
- I am using Macbook Pro 14" M1 Pro (16GB, 512SSD), MacOS 12.3
- Optimized battery charging is turned off

Charge on my Macbook Pro 14" M1 Pro is not being limited using the AlDente Free version.

**To Reproduce**
1. Discharge battery to 70%
2. Set battery limit to 80%.
3. Plug-in the charger into MacBook
4. MacBook keeps charging until battery is 100%

**Expected behavior**
MacBook battery charge should stay around ~ 80%.

Screenshots
[image]

[image]

**MacBook (please complete the following information):**

MacBookPro18,3 - MacBook Pro 14" 16GB RAM, 512GB SSD
MacOS 12.3
AlDente Free 1.15.2

radovansurlak, 2022-03-25:
I have tried reinstalling the app and helper. I have noticed that when I plug-in my charger, it shown battery is not charging briefly, but then it disappears. Maybe it doesn't limit the charge after some point after the new OS update?

MatthiasKerbl, 2022-03-25:
> I have tried reinstalling the app and helper. I have noticed that when I plug-in my charger, it shown battery is not charging briefly, but then it disappears. Maybe it doesn't limit the charge after some point after the new OS update?

Hi @radovansurlak ,

This behavior is probably caused by the macOS update.

Please try the following to resolve this issue:

1. Shut Down your MacBook completely.
2. Unplug the power supply from the socket.
3. Wait a couple of minutes.
4. Restart your MacBook and plug it in again.

Does it work again?

Cheers,
Matthias

radovansurlak, 2022-03-25:
Hi again @MatthiasKerbl.

Here's what I have done (my battery was at 39%, unplugged)
1. Reinstalled AlDente
2. Set charge limit to 35% 
3. Shut down the MacBook unplugged
4. Waited for a few minutes
5. Booted the MacBook up
6. Plugged in the charger
7. The battery status shows "Battery is not charging" for 2 seconds
8. Battery status then shows "Power Source: Power Adapter"
9. Battery percentage keeps rising (at 46% at the moment of writing this)

Unfortunately it seems to be still happening for me.

I am not sure if it would have worked with the older MacOS as I have got this MacBook fresh with the 12.3

vishwasvogga, 2022-04-10:
been there, same issue. but i installed al dente pro version 1.15.2 after upgrading to montero OS 12.3. None of the features working. already opened a ticket #653

ospop, 2022-04-14:
same here... Sailing mode is not working, 'power source: power adapter'

ospop, 2022-04-14:
> same here... Sailing mode is not working, 'power source: power adapter' Also, it keeps charging past the limit I've set. This is happening since I upgraded to PRO version, it used to work properly before.
> 
> * Macbook Air M1

saltage, 2022-04-15:
I am having the same issue. My battery just continues charging past the set limit and when I select discharge, it intermittently switches between charging and discharging.

kmcgraw321, 2022-04-19:
I am seeing similar behavior with AlDente 1.16.1. I am running on a 2019 16" MBP. I suspect the problem is related to the most recent OS patch as AlDente Classic 2.1.2 was working fine prior to the OS update. After the update it quit working. This prompted me to install AlDente 1.16.1 to pick up any fixes. My battery charge level is currently higher than the AlDente setting. After installing the helper the battery status changes from charging to not charging but only for a few seconds. The status changes back to charging and the charge level is slowing increasing.

MatthiasKerbl, 2022-04-19:
> I am seeing similar behavior with AlDente 1.16.1. I am running on a 2019 16" MBP. I suspect the problem is related to the most recent OS patch as AlDente Classic 2.1.2 was working fine prior to the [...]

Hi @kmcgraw321,

Do you have "Optimized battery charging" in Settings disabled?

radovansurlak, 2022-04-20:
@MatthiasKerbl I have installed AlDente 1.16.1, turned off optimized charging.
Charge limit now works as expected, thank you for your work!

kmcgraw321, 2022-04-20:
> > I am seeing similar behavior with AlDente 1.16.1. I am running on a 2019 16" MBP. I suspect the problem is related to the most recent OS patch as AlDente Classic 2.1.2 was working fine prior to the OS update. After the update it quit working. This prompted me to install AlDente 1.16.1 to pick up any fixes. My battery charge level is currently higher than the AlDente setting. After installing the helper the battery status changes from charging to not charging but only for a few seconds. The status changes back to charging and the charge level is slowing increasing.
> 
> Hi @kmcgraw321,
> 
> Do you have "Optimized battery charging" in macOS Settings disabled?

Yes, I have "Optimized battery charging" disabled. Also an update to my status. Reading responses to other post I tried disconnecting the power cord and power cycling the laptop. This seems to have resolved my problem. It now consistently indicates that the battery is not charging and the charge level is slowly dropping to my set point.