AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #629: Lock Screen Battery Not Same as Unlocked

SkyBlue1903, 2022-03-14:
Hey, can someone please tell be between this differences:
On lock screen:
[image]
When unlocked:
[image]
Is AlDente supposed to be working correctly? I set charging limits to 50%. This thing kinda suspicious. Hope for the answer, thanks!

andrewcrook, 2022-03-22:
I know AlDente doesn’t have control over that screen.
 I believe it maybe the norm.

MatthiasKerbl, 2022-03-23:
Hi @SkyBlue1903,

this is just an inconsistency in macOS. AlDente is still doing its thing in the background.

Cheers,
Matthias

closed 2022-03-23 by MatthiasKerbl