AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #630: Improve Heat Protection: Active cooling while charging

nheeren, 2022-03-16:
Maybe this applies only to the M1 Macs, but I noticed that the battery of my MacBook Pro becomes quite hot while charging. The cooling system of this new MacBook design is very efficient and also the air inlet passes the batteries, cooling them effectively. However, because the Mac is so efficient, the fans actually almost never activate. 

So I tried to activate the fans during charging today and the temperature reduction at a fan speed of 50% was considerable. I used iStat Menus to control the fans and measure temperature.

[image]

So since you already read the temperature sensors in your Heat Protection feature, I was thinking it might be quite simple to control the fans as well. Please do not only implement an on/off option only, but a more fine-grained control to keep noise at bay ;)

nheeren, 2022-03-18:
For those that are also using iStats, I just realized that there is already such a functionality...

[image]

andrewcrook, 2022-03-19:
> For those that are also using iStats, I just realized that there is already such a functionality...

> do not only implement an on/off option only, but a more fine-grained control to keep noise at bay

Can iStats be set so that it has to **be charging and over a certain threshold.**

Perhaps AlDente could do this?
but still have a point where charging is stopped to fallback on.

nheeren, 2022-03-20:
> Can iStats be set so that it has to **be charging and over a certain threshold.**

The setting can be refined a bit more:
[image]

andrewcrook, 2022-03-21:
@nheeren  Do we know how to get the battery temp in istats ?

nheeren, 2022-03-23:
@andrewcrook Not sure if I understand your question correctly, but it is one of the available sensors. Maybe not all Macbooks have such a sensor. I am on am MBP 16" 2021.

[image]

andrewcrook, 2022-03-29:
> @andrewcrook Not sure if I understand your question correctly, but it is one of the available sensors. Maybe not all Macbooks have such a sensor. I am on am MBP 16" 2021.
> 

Yeah somingthing isnt right?
M1 13” MBP 2020

[image]

nheeren, 2022-03-29:
Interesting. When you activate Heat Protection in Al Dente, does it display a temperature? @andrewcrook

[image]

andrewcrook, 2022-03-29:
@nheeren 

Yes and it seems to be correct (or what I would expect). However, doesn’t match any of the other sensors I think it must be missing from from iStats 

[image]

Battery etc does show up in Macs Fan Control.

[image]

andrewcrook, 2022-03-30:
@nheeren just to let you know the battery issue in istats for 13” M1 is now fixed in 6.62.

But it would still be great to have a fan feature in Al Dente

alexbaramilis, 2022-05-09:
That would be a very neat feature indeed.

I have an issue with heat protection and an improvement that I described in #675, but if we can have the fans turning on as well, that would be even better!

MatthiasKerbl, 2023-12-06:
> The heat protection feature is very good but sometimes I need to chage the battery up to max capacity as much as quickly. In that case I need to disable that feature or open another app(in my case it is tg pro) to set the fans max rpm as manually.
> 
> I would be happy if can the Al Dente can control the fans when needed. Because I use the TG Pro only for that case, I dont need it in my normal usage.

Hi @SevcanAlkan,

including the fans is something we might include in a future update. For now, if you want to charge the battery to 100% quickly, you can use the Top Up feature. During Top Up, the Heat Protection feature is ignored automatically.

Best,
Matthias

andrewcrook, 2023-12-06:
@MatthiasKerbl 

I was thinking,
Do any of these Mac Fan Control Apps use AppleScript and/or Shortcuts?

If so adding some configurations to trigger AppleScript and or Shortcuts on events and passing values might be a great way  to expand some of the functionality and opens it up for all assorts of integrations and new features.

MatthiasKerbl, 2023-12-07:
Hi @andrewcrook,

> Do any of these Mac Fan Control Apps use AppleScript and/or Shortcuts?

I am not sure since I have not done a deep dive on Mac Fan Control apps before.

> If so adding some AlDente configurations to trigger AppleScript and or Shortcuts on events and passing values might be a great way to expand some of the functionality and opens it up for all sorts of integrations and new features.

This should already be possible from our end through the Apple Shortcuts integration of AlDente.

Long term, we are going to add fan control functionality to AlDente Pro, we are just not focusing on it right now.

Best,
Matthias