AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #634: Aldente stopped working on MBA 2020 M1, battery charges above set threshold.

panpietrek, 2022-03-19:
Aldente stopped working (battery keeps charging above set threshold) on my MBA M1 2020.
The following appears in the logs while starting it up:

[image]

 - MacBook model MacBook Air M1 2020 10.1
 - macOS version 12.3 
 - AlDente Pro or AlDente Free? Free
 - AlDente Version? 1.15.2
 - List of all enabled/disabled settings in AlDente:
Pretty much OOTB, except for charging limit set to 80%

I tried reinstalling the helper and rebooting, then reinstalled the entire app - no good.
It was fine last week, until I upgraded to 12.3 two days ago, so I think macOS update might be the culprit here.

Thanks.

MatthiasKerbl, 2022-03-22:
Hi @panpietrek,

This behavior is probably caused by the macOS update.
The errors in the logs can be ignored since they are only valid for Intel MacBooks.

Please try the following to resolve this issue:

1. Shut Down your MacBook completely.
2. Unplug the power supply from the socket.
3. Wait a couple of minutes. 
4. Restart your MacBook and plug it in again.

Does it work again?

Cheers,
Matthias

panpietrek, 2022-03-22:
Holy macaroni! It worked!
Aldente stops charging the way it should now. Thanks!

closed 2022-03-22 by panpietrek