AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #636: [Feature Request] Save and import export battery history and show other information such as health

andrewcrook, 2022-03-19:
I think it would be great to be able to store and view battery history  e.g 

**So the main data would be ..** 

- cycles
- macOS battery health status 

**You could also add**

- calibrations
- full charge capacity  (mAh)
- temperature of battery?
- see common ranges/averages of discharge charging from and when charged 
   so you can see when stuck in a habit and when to add some variance.

**viewing:**
Would also be useful to see charts and able to compare different data types

**import and export:**
via CSV file
backup
import common formates from third party eg coconutBattery and BatteryHealth 3 etc and restore back up

**other information to view in UI:**
- battery age (days)
- serial 
- manufacture date

Note I think that historical information would help other features finding optimal schedules for calibrations such as 

- [Automatic calibration](https://github.com/davidwernhart/AlDente/issues/572) #572 
- [Ability to schedule Calibration ](https://github.com/davidwernhart/AlDente-Pro-Bug-Tracking/issues/29)

itssiggi, 2022-05-01:
Yes automatically saving battery health & cycles daily and viewing/exporting the data later on would be really nice!

Egregius, 2023-09-22:
Any news on this? Would love to have the data stored in a MySQL database for further analysis etc.

MatthiasKerbl, 2023-09-22:
> Any news on this? Would love to have the data stored in a MySQL database for further analysis etc.

Hi @Egregius,

It's on our internal roadmap but not a priority at the moment.

Best,

Matthias

andrewcrook, 2023-09-23:
> data stored in a MySQL database

I would recommend using native SQLite/iCloud support. But there could easily be an automated way to export to third-party databases via scripting. The things is, people use many different types of data storage these days.

Blackspell01, 2025-01-31:
Just curious, if I reset my Mac, everything is gone for now? Or where does coconutBattery save the history?

MatthiasKerbl, 2025-02-06:
Hi everyone,

The latest version of AlDente (1.30) has a new Dashboard feature that visualizes important battery data.

Best,
Matthias

MatthiasKerbl, 2026-10-06:
Hi @andrewcrook,

With AlDente 1.30, we added the Dashboard, which visualizes the most important battery data over time, such as battery level, cycles, maximum capacity, and power consumption. Since AlDente 1.35, the Battery Health Stats and the Battery Cycles graph are also available in the free version.

I will close this issue for now. The export/import part of this request is tracked in #1103.

Best,
Matthias

closed 2026-10-06 by MatthiasKerbl