AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #637: [Feature Request]  command line interface

andrewcrook, 2022-03-19:
**command line interface**

- for people who prefer shell 
- for scripting and automation

**features**  

- view information
- view settings
- change settings
- start/stop charging
- start/stop discharge
- enable/disable AlDente
- top up
- start/stop calibration
- get calibration status 
- import/export data
    #636 

**options** 
--json (-j) output JSON for scripting etc.

infused-kim, 2022-07-02:
+1 on this. 

It would allow us to create raycast and Alfred workflows. 

Could also be another pro feature that could be very compelling.

MatthiasKerbl, 2023-03-13:
Hi everyone,

Not yet full CLI compatibility but AlDente Pro is compatible with Apple Shortcuts for a while now and since some Shortcuts can be run from the command line, you can already now control AlDente Pro through the command line to some degree.
You can read more about this topic here:

- https://apphousekitchen.com/feature-explanation-shortcuts/
- https://support.apple.com/guide/shortcuts-mac/run-shortcuts-from-the-command-line-apd455c82f02/6.0/mac/13.0

Cheers,
Matthias

kzssc, 2024-11-14:
sorry for digging - also interested in an alfred workflow!