AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #638: [Feature Request] AppleScript and Shortcuts Interfaces.

andrewcrook, 2022-03-19:
expanding  on #637 

Also add AppleScript and Shortcuts interfaces to control and alter settings.

ConnorBosman, 2024-06-27:
Expanding on this, what about maybe using the Mac URL shortcut to automatically trigger shortcuts? That would be great imo, for example by automatically triggering a shortcut upon a specific battery percentage.

https://support.apple.com/guide/shortcuts-mac/run-a-shortcut-from-a-url-apd624386f42/mac

andrewcrook, 2024-06-27:
@ConnorBosman do you mean so you can add hooks that AlDente can call shortcut . 
So it can start shortcuts on certain events?

ConnorBosman, 2024-06-27:
Exactly. The current tasks is nice, but it’s not very useful. I’d for example like to turn on a fan automatically when my battery gets too hot.