AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #640: AlDente preventing Sleep

noodlzsynthetase, 2022-03-26:
Describe the bug
AlDente helper is stopping my computer from going to sleep both on the lock screen/ desktop. I read a similar thread and tried updating and restarting the mac, and reinstalled it as well but it doesnt help. After I force quit aldente and aldente the computer again goes to sleep.

To Reproduce
Steps to reproduce the behavior:

Download and install AlDente
Turn it on and set a limit
Computer does not  go to sleep
Expected behavior
As I have set display off in 3 mins, it normally should go to sleep, which it does after closing Aldente

MacBook:

MacBook model: MacBook Pro (13-inch, 2021) - M1
macOS version: 12.3
AlDente Premium
AlDente Version: 1.15.2
List of all enabled/disabled settings in AlDente: All defaults
[image]

tchpanda, 2022-03-29:
Exactly the symptoms I'm experiencing.  I only purchased this software today.

Macbook Pro 16" (2021) - M1
macOS Monterey 12.3
AlDente Premium
AlDente Version: 1.15.2
List of all enabled/disabled settings in AlDente: Sailing Mode ON, Heat Protection OFF, Stop charging when app closed/sleep: OFF

tchpanda, 2022-03-29:
~ % pmset -g
System-wide power settings:
 SleepDisabled		1
Currently in use:
 standby              1
 Sleep On Power Button 1
 hibernatefile        /var/vm/sleepimage
 powernap             1
 networkoversleep     0
 disksleep            10
 sleep                1 (sleep prevented by powerd, bluetoothd)
 hibernatemode        3
 ttyskeepawake        1
 displaysleep         1
 tcpkeepalive         1
 lowpowermode         0
 womp                 1

tchpanda, 2022-03-29:
[aldentepro-debug-2022-03-29-21:07:05.pasta.txt](https://github.com/davidwernhart/AlDente/files/8370720/aldentepro-debug-2022-03-29-21.07.05.pasta.txt)

MatthiasKerbl, 2022-03-29:
Hi @noodlzsynthetase and @tchpanda,

do you have the feature "Disable Sleep until Charge Limit" enabled?

[image]
When you click on the Apple icon in the upper left corner, is the sleep icon greyed out like that?

Cheers,
Matthias

tchpanda, 2022-03-29:
This machine spends a LOT of time in 'clamshell' mode attached to a large monitor via USB-C.  

Not sure if this is a 100% supported option but the documentation seemed to imply the paid version of this software addressed those issues.

tchpanda, 2022-03-29:
I removed and reinstalled the helper and now the 'Sleep' option is enabled in the Apple menu, so...possible progress?

noodlzsynthetase, 2022-03-29:
> Hi @noodlzsynthetase and @tchpanda,
> 
> do you have the feature "Disable Sleep until Charge Limit" enabled?
> 
> [image]
> 
> When you click on the Apple icon in the upper left corner, is the sleep icon greyed out like that?
> Cheers, Matthias

It isnt greyed out currently because I restarted the computer, but it was previously. No I do not have the "Disable Sleep until Charge Limit"  enabled.

andrewcrook, 2022-03-29:
I am not getting the same issue and I cannot think why.

M1 13" MBP 2020
macOS 12.3
AlDente Pro 1.15.2

I have tried "Disable Sleep until Charge Limit” enabled with no issue.
A part from keeping the external display a wake. I think this should be a separate option.

Disable macOSs battery optimisation which I expect you have done.
I have heard that you need to restart the machine after installation to resolve some issues. 

> MacBook model: MacBook Pro (13-inch, 2021) - M1 

@noodlzsynthetase Do you mean 2020? 
I don't think the 13” had a hardware refresh last year since 2020.

andrewcrook, 2022-03-29:
@tchpanda 

> This machine spends a LOT of time in 'clamshell' mode attached to a large monitor via USB-C.

That’s how I use mine as well. 
The only issue I have is I gain 1pc a day or two I think this is when I have powered down and turn mains off. 
There is some time between power on, booted and AlDente loading where it might charge a bit. I guess really I should keep in sleep.

Brax68, 2022-03-29:
Same issue on my 14" 2021 M1 MacBook Pro, I uninstalled and it fixed it. I was using the free version and it didn't have don't sleep until done charging enabled. The other odd thing is that it prevents sleep even in the boot manager... Went to the genius bar and he even tried closing it during the boot up selector, it didn't sleep. The helper is changing some sort of firmware and is causing issues.

noodlzsynthetase, 2022-03-30:
> I am not getting the same issue and I cannot think why.
>
> M1 13" MBP 2020 macOS 12.3 AlDente Pro 1.15.2
>
> I have tried "Disable Sleep until Charge Limit” enabled with no issue. A part from keeping the [...]

yes I do, sorry it is MBP 13' 2020 M1.

Brax68, 2022-03-30:
I have been using it for a while... Since 12.2 came out and didn't have any issues until 12.2.1. I had my MacBook not turn off when I closed the lid, resolved by itself after a restart. I have done everything and haven't purchased pro, but still get issues. I am not reinstalling until there is some sort of fix.

tchpanda, 2022-03-30:
I just can't tell if this is working as intended.  The machine appears to sleep now, but battery status does not change from 81%.  I have been using the machine all morning.  I have 'Sailing mode' enabled and set to 50%.

[image]

tchpanda, 2022-03-31:
I'm going to go with 'this is not working as intended'.  Several hours on:

[image]

MatthiasKerbl, 2022-03-31:
> I just can't tell if this is working as intended. The machine appears to sleep now, but battery status does not change from 81%. I have been using the machine all morning. I have 'Sailing mode' enabled and set to 50%. Charge limit is set to 75%.
> 
> [image]

Hi @tchpanda,

Please take a look at our detailed feature explanations on our blog:
Sailing Mode: https://apphousekitchen.com/feature-explanation-sailing-mode/
Disable Sleep until Charge Limit: https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/

Cheers,
Matthias

andrewcrook, 2022-03-31:
> I just can't tell if this is working as intended. The machine appears to sleep now, but battery status does not change from 81%. I have been using the machine all morning. I have 'Sailing mode' enabled and set to 50%. Charge limit is set to 75%.
> 
> [image]

A according to that charging is paused,  not using battery and using mains power.  It should say on around 81pc
I would expect the batter to go down if you’re unplugged from mains,  with sailing I wouldn’t expect charging to start until below 50% and on mains power.

Also don't forget some external monitors might be supplying power to the Mac via USB-C.

MmadJj, 2022-04-03:
AlDente Free Version works as expected on my M1 16" Macbook Pro. However the laptop pulls 2-3W constantly in sleep mode. When i uninstall AlDente and go into sleep Mode the Macbook uses 0W.

davidgirard, 2022-05-03:
I just did last night, w/ 1.16.1 (full/paid). Same symptoms. MBP never went to sleep. Just checked, sleep option is greyed out.

MatthiasKerbl, 2022-10-04:
Hi @noodlzsynthetase,

Do you still experience this issue with our newest version AlDente 1.18? https://github.com/davidwernhart/AlDente/releases
Otherwise, I will close this issue.

Cheers,
Matthias

MatthiasKerbl, 2022-10-31:
Hi @noodlzsynthetase,

I will close this issue for now.
If you have any updates on this let me know.

Cheers,
Matthias

closed 2022-10-31 by MatthiasKerbl

vgiannadakis, 2023-04-20:
This happened to me as well, here's my info:

```
Model Name:              MacBook Pro
Model Identifier:        MacBookPro18,3
Model Number:            Z15G001X7LL/A
Chip:                    Apple M1 Pro
Total Number of Cores:   10 (8 performance and 2 efficiency)
Memory:                  32 GB
System Firmware Version: 8422.100.650
OS Loader Version:       8422.100.650
```

AlDente Pro 1.21.4

```
~ % pmset -g
System-wide power settings:
 SleepDisabled		0
 DestroyFVKeyOnStandby		0
Currently in use:
 standby              1
 Sleep On Power Button 1
 hibernatefile        /var/vm/sleepimage
 powernap             1
 networkoversleep     0
 disksleep            0
 sleep                0 (sleep prevented by com.apphousekitchen.aldente-pro, coreaudiod, coreaudiod, powerd, bluetoothd)
 hibernatemode        3
 ttyskeepawake        1
 displaysleep         20 (display sleep prevented by com.apphousekitchen.aldente-pro)
 tcpkeepalive         1
 lowpowermode         0
 womp                 0
```

I was able to remedy by reinstalling the helper application. After doing that:

```
giannae@bcd0741d0ad3 ~ % pmset -g
System-wide power settings:
 SleepDisabled		0
 DestroyFVKeyOnStandby		0
Currently in use:
 standby              1
 Sleep On Power Button 1
 hibernatefile        /var/vm/sleepimage
 powernap             1
 networkoversleep     0
 disksleep            0
 sleep                0 (sleep prevented by coreaudiod, coreaudiod, powerd, bluetoothd)
 hibernatemode        3
 ttyskeepawake        1
 displaysleep         20 (display sleep prevented by <redacted>)
 tcpkeepalive         1
 lowpowermode         0
 womp                 0
```

StefCobe, 2023-07-10:
Sometimes it happens to me too. I solved this by reinstalling the helper (Al Dente -> General -> Reinstall Helper).

Razer0123, 2025-01-15:
I think options to block charging while sleeping are causing sleep issues

xin4869, 2025-02-23:
Hello, I am still experiencing this issue in ALDente Pro 1.31.1. Please help

xin4869, 2025-02-23:
even after reinstall helper, the issue persists. the 'sleep'button work only right after reinstall helper application. Soon 'sleep'button goes back to non-selective mode