AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #646: Charge limit resets itself to 100%

pkowal-x, 2022-04-02:
It happened already three times in the last 30 days. For unknown reason I notice, that the battery is fully charged, to 100%. I set AlDente to 79%, so this is strange. When I open AlDente it shows, that the charge limit is set to 100%. Then I set it again to 79% and... in few days it is 100% again. I expect it to stay at desired percentage.

Intel MacBook Pro 13".
MacOS Monterey (currently 12.3.1)
AlDente Free 1.15.1
Intel mode on

MatthiasKerbl, 2022-04-05:
> It happened already three times in the last 30 days. For unknown reason I notice, that the battery is fully charged, to 100%. I set AlDente to 79%, so this is strange. When I open AlDente it shows, [...]

Hi @pkowal-x,

Please update to AlDente 1.15.2 and let us know if you still encounter this bug.

Cheers,
Matthias

MatthiasKerbl, 2022-04-05:
Hi @pkowal-x and @messin9,

Does the slider of AlDente go to 100% after a while or does only your MacBook charge to 100%?

Cheers,
Matthias

Muri96, 2022-04-06:
Same problem here ... charging to 100% although I set 60%. The slider stays at 60%

Also the icon jumps from charge to pause - see video:

https://user-images.githubusercontent.com/103127120/161997697-ddec369f-08d0-4a92-910c-3863e7a83eaf.mov

.

MatthiasKerbl, 2022-04-06:
> Same problem here ... charging to 100% although I set 60%. The slider stays at 60%. Version Ardente Pro 1.15.2
> 
> Also the icon jumps from charge to pause - see video:
> 
>  Bildschirmaufnahme.mov 
> .

Hi @Muri96,

Please open a new issue here on Github and fill out the questions for a bug report, since it seems to be a different issue then @pkowal-x was reporting.

Thanks,
Matthias

chaoqunsoft, 2022-04-08:
Same here, charge limit is not work on 12.3.1
M1 silicon

MatthiasKerbl, 2022-04-08:
> Same here, version 1.15.2 , charge limit is not work on 12.3.1 M1 silicon

Hi @chaoqunsoft,

Did you do a macOS update lately?
If yes, this might be the reason for your issue.

Please try the following to resolve this issue:

1. Shut Down your MacBook completely.
2. Unplug the power supply from the socket.
3. Wait a couple of minutes.
4. Restart your MacBook and plug it in again.

Does it work again?

Cheers,
Matthias

Blinks44, 2022-04-08:
Same here, after macOS 12.3.1 update, aldente's charge limit doesn't work anymore. Trying to reset helper doesn't help either.

chaoqunsoft, 2022-04-10:
> Hi @chaoqunsoft,
>
> Did you do a macOS update lately? If yes, this might be the reason for your issue.
>
> Please try the following to resolve this issue:
>
> 1. Shut Down your MacBook completely.
> 2. Unplug [...]

It's worked. thanks buddy.

DarthGigi, 2022-07-20:
I'm having the same issue as OP, AlDente resets the charge limit to 100%, often after a reboot but not necessarily.
I'm on AlDente Version 1.17.1 and macOS Monterey version 12.5 (had this issue in previous versions too).

MatthiasKerbl, 2022-10-31:
Hi @DarthGigi and @pkowal-x,

Do you still experience this issue with our newest version AlDente Pro 1.19.1?
Over all those months, I did not experience this issue even once myself.
Any ideas on how to reproduce it if it still occurs?

Cheers,
Matthias

pkowal-x, 2022-10-31:
Hmm... I must say I encountered it suddenly today. The battery was at about 60%, my AlDente limit is 88%, and since I just got the notification that you wrote I look at my battery meter and I see 95%, power source: adapter, battery not charging. Not sure why it went up to 95% (?).

M1 Pro MacBook Pro 16".
MacOS Monterey 12.6.1
AlDente Free 1.19.1

pkowal-x, 2022-10-31:
Interesting - now when I check it it says 96%
[image]

[image]

pkowal-x, 2022-10-31:
And now... 97%. Is there any possibility, that the app works differently in free and pro version? I can't change any setting in the "charge" tab, so it is default. I only have it set at 88% for last months, and the slider stays there. 

[image]

[image]

MatthiasKerbl, 2022-11-01:
Hi @pkowal-x,

If I understand correctly, your current issue is that the Charge Limiter does not work?
Did you do a macOS update recently?

Cheers,
Matthias

MatthiasKerbl, 2022-11-01:
Please try the following:

> AlDente stops working after the macOS update!
> 
> Unfortunately, AlDente temporarily does not work after a macOS update for a small number of users.
> 
> To get AlDente to work again, please try the following steps which have worked for many users before:
> 
> 1. Unplug and discharge your MacBook to less than your current charge limit in AlDente
> 2. Turn off your MacBook and close the lid for a couple of minutes
> 3. Unplug the power brick from the outlet and wait a couple of minutes
> 4. Turn on your MacBook and plug it in again
> 5. Check if it stops charging at the charge limit set in AlDente
> 
> We are aware of this bug for a while now but were so far not able to fix it. However, we will keep trying.

MatthiasKerbl, 2022-11-24:
Hi @pkowal-x,

Any updates on this?
Did you try my suggestions above?

Cheers,
Matthias

pkowal-x, 2022-11-28:
Hey Matthias! It seems like it is working fine now, thank you :).

closed 2022-11-28 by pkowal-x