AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #647: Discharge function stopped working on Intel Mac after update to 12.3.1

Archer785, 2022-04-05:
**Describe the bug**
My limit is set to 70%, I hit discharge button and it doesnt do anything.

**MacBook (please complete the following information):**
 - MacBook Pro 13-inch, 2019
 - macOS version 12.3.1, freshly updated
 - AlDente Free
 - AlDente Version - 1.15.2
 - List of all enabled/disabled settings in AlDente: Disabled Intel Mode

sameh0, 2022-04-05:
I'm getting the same behavior on Apple Silicon as well

MatthiasKerbl, 2022-04-05:
Hi @Archer785 and @sameh0,

I just did the mac OS update on my 2016 15" MacBook Pro and Discharge works without any issues. Mac OS just needs a couple of seconds to show the changes.
Does discharge still not work for you?
Did you try a restart of your MacBook?

Cheers,
Matthias

sameh0, 2022-04-05:
Shutting the mac down for some time fixed the issues . 
Thank you @MatthiasKerbl

Archer785, 2022-04-06:
I have shut the device down overnight for good measure and it fixed the issue, thanks @MatthiasKerbl !

closed 2022-04-06 by Archer785