AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #648: MacBook ignores charging-limit

Muri96, 2022-04-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

Muri96, 2022-04-06:
> **Describe the bug** A clear and concise description of what the bug is.
>
> **To Reproduce** Steps to reproduce the behavior:
>
> 1. Go to '...'
> 2. Click on '....'
> 3. Scroll down to '....'
> 4. See error [...]

MatthiasKerbl, 2022-04-06:
Hi @Muri96,

Did you do a macOS update lately?

Muri96, 2022-04-06:
yes, a view days ago

MatthiasKerbl, 2022-04-07:
Hi @Muri96,

This is probably the reason for the issues you encounter.

Please try the following to resolve this issue:

1. Shut Down your MacBook completely.
2. Unplug the power supply from the socket.
3. Wait a couple of minutes.
4. Restart your MacBook and plug it in again.

Does it work again?

Cheers,
Matthias

closed 2022-04-07 by Muri96