AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #650: Menu Bar - icon Not visible in MacBookPro M1 ( 14inch-2021 ) due to camera area.

openui54u, 2022-04-08:
**Describe the bug**
The icon of AlDente is not place correctly in Menu bar. There is spacing but still not visible due to to camera -area in the MacBookPro M1 ( 14"-2021 ) .

**To Reproduce**
Use a MacBookPro 14"-2021 and have AlDente active/installed.
Copy with screen copy area incl. behind the camera area and see that the icon is present but not reachable/visible: with some good guessing you can get it though....

**Expected behavior**
Expected is the icon to be visible left or right from the non-visible area.

**Screenshots**
(Internal) screen copy show icon behind the camera-nodge
[image]

Ext.Camera screenphoto showing camera-nodge:
[image]

**MacBook (please complete the following information):**
 - MacBook model MacBookPro M1 Pro 14"-2021   
 - macOS version 12.3.1 Monterey
 - AlDente Free?
 - AlDente Version 1.15.2

nheeren, 2022-04-11:
Exactly why I proposed #547

kentzh1, 2022-04-11:
bartender4 can help you

openui54u, 2022-04-11:
I tried some reinstall en restart. No luck. But inhad some old programm icons ( just upgraded to new laptop and M1 chip ) that could be removed and so after removing 2 ivons from menubar the aldente icon appeared. So no real solution but I am currently fine ...

MatthiasKerbl, 2022-11-05:
Hi @madler,

Do you still experience the AlDente icon in the middle of the menu bar with our newest version AlDente 1.19.2?

Cheers,
Matthias

openui54u, 2022-11-05:
Hi Thank you for asking.
Short: yes

I actually posponed the update few times. Just checked this respons with 1.19.1 
Than I updated.
I tested again ( adding the Siri icon to the menu bar as a test pushing AlDente off the viewable area…before)… and it still is.

To bad. This (crazy) notch in the middle is in the way 😀 of viewing ( in this case ) AlDente icon..

Thank you for your efforts. Do you think it’s something of AlDente or that the OS Monterey should fix that?

Edwin Glaser

MatthiasKerbl, 2022-11-05:
Hi @openui54u,

Thanks for the quick reply.
Could you please share a screenshot of the menubar area?
I would like to see if there is still an unusually big gap between the AlDente icon and the next icon or if it just gets pushed under the notch.

Cheers,
Matthias

openui54u, 2022-11-05:
Can tou see it? Sorry pressed semt button bedore typing explanatiin.
The aldente icon is om the left sode of the row if icons nright next to the notch as you can see.
When i i.e. add a siri icon to the menubar the aldente will be pushed under the notch.

Met vriendelijke groeten,

Edwin Glaser

Verstuurd vanaf mijn iPhone

openui54u, 2022-11-05:
Oh my, I typed this on my iPhone….I did not have my reading glasses on… .sorry for the typo’s Hope you understand 😀 

So: the icon of AlDente gets pushed under the notch when adding another icon in the right side ( here I do it with Siri as example )
I filmed a bit from above so you hopefully can see the black notch against the dark grey background  ( it is nighttime here, so the colors are now in Dark mode 😀   If needed to I can switch to light mode… but the idea and space is clear: I do believe with the size of icons this one extra icon should JUST have fitted next to it.. And the funny thing is…the icon does NOT appear to the left side of the notch… it s just behind it and not a bit, quite a big part. 

Edwin

openui54u, 2022-11-05:
Met vriendelijke groeten,

Edwin Glaser

Verstuurd vanaf mijn iPhone

openui54u, 2022-11-05:
Video os bouncing back. Here jsut the photo. [image0.jpeg]

Met vriendelijke groeten,

Edwin Glaser

Verstuurd vanaf mijn iPhone

openui54u, 2022-11-05:
[image0.jpeg][image1.jpeg]

Met vriendelijke groeten,

Edwin Glaser

Verstuurd vanaf mijn iPhone

openui54u, 2022-11-06:
[image0.jpeg]

Met vriendelijke groeten,

Edwin Glaser

Verstuurd vanaf mijn iPhone

MatthiasKerbl, 2022-11-06:
Hi @openui54u,

Unfortunately, I am not able to see any of the pictures/videos you shared. If you shared them as an attachment with an email, they don't show here on GitHub. Please try to upload them directly here in GitHub again so that I can assess the situation better.

Cheers,
Matthias

openui54u, 2022-11-06:
herewith

https://user-images.githubusercontent.com/24986693/200179610-e77965e2-de53-47a3-ace0-0d02a2f95ed7.mp4

[image]
[image]
[image]
[image]
[image]
[image]

MatthiasKerbl, 2022-11-06:
Thanks for the video and pictures.
As far as I can see, the issue is that macOS is pushing the AlDente icon behind the notch because of a lack of space next to it. This is unfortunately an issue from macOS and you can circumvent it by using a menubar icon manager such as Bartender.  I am not able to see any bug or issue coming from AlDente here.

Cheers,
Matthias

openui54u, 2022-11-06:
OK. thanks for your time. Please have a look at the icon being OVER the menu options of another program…. 😀 also Mac OS failure?

MatthiasKerbl, 2022-11-06:
> Please have a look at the icon being OVER the menu options of another program…. 😀 also Mac OS failure?

This behavior seems weird and I have never seen it before. It is hard to say if it is an AlDente, macOS, or third-party app issue. Is it only happening with one particular program?

MatthiasKerbl, 2022-12-24:
Hi @openui54u,

Since the original issue seems to be resolved, I will close this issue.
If you experience the overlayed AlDente icon regularly, please open a new issue.

Cheers,
Matthias

closed 2022-12-24 by MatthiasKerbl