AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #652: Sailing Mode will not sail to 75%, but is going lower

KAMIKAZEUA, 2022-04-09:
**Describe the bug**
Sailing mode is not working 74%

**To Reproduce**
Steps to reproduce the behavior:
Enable it, use MacBook..

**Expected behavior**
Drop to 75% and started to charge

**Screenshots**

[image]
[image]
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook Pro (14-inch, 2021)
 - macOS version 12.3.1(always using latest)
 - AlDente Pro
 - AlDente 1.15.2
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Other app, like coconutBattery showing that it’s not charging and it’s real % is 74.32% and that it's not charging, so it's still trying to discharge battery.

KAMIKAZEUA, 2022-04-09:
And after some time it's going lower, but system and AlDente showing that it's has higher value of battery charge: 

[image]

This is really strange as for me..

MatthiasKerbl, 2022-04-09:
Hi @KAMIKAZEUA,

From what I can see, Sailing Mode works as intended for you. Please read the following explanation of the Sailing Mode completely: https://apphousekitchen.com/feature-explanation-sailing-mode/

Regarding the battery percentage: MacOS shows you a different percentage than the battery actually has. This is due to various reasons. For simplicity, AlDente currently uses the macOS battery percentage. However, in the next update of AlDente, you will have the option to switch to the hardware battery percentage in AlDente as well. Which is the same one the other apps show you.

Cheers,
Matthias

KAMIKAZEUA, 2022-04-09:
Hey @MatthiasKerbl but using real values is the only right way for battery, since it's sailing on a wrong/non optimal % range for li-ion batteries(real 75-80%).

When this new version will be available?

MatthiasKerbl, 2022-04-09:
Hi @KAMIKAZEUA,

the next update with the hardware battery percentage will be released in the coming weeks of April.

Cheers,
Matthias

KAMIKAZEUA, 2022-04-11:
Hey @MatthiasKerbl that's good. But I really wondering now which way the the right way..
Interesting how Apple did this trick with real % of battery and those from Settings and why...

So, currently this app works using system value and as results typical settings is 80% limit and -5% for sailing mode.
As result for real values, this will be: 

-  75% system = ~71.45% real value:  [image]

-  80% system = ~78.95 real value:   [image]

KAMIKAZEUA, 2022-04-13:
Thanks for quick fix and update just released today :) 

I'll test properly, but so far what I can see 2 bugs.
 1.  it's not reading % precisely: 

[image]

Should be some option like from other app(Wattagio):

[image]
 
2. It tells that it's Disabled, when it's Enabled :)

[image]

_P.s. have no idea why you really rushing things here by closing ticket, why not just wait for update from user, wait for feedback after new release, confirm it's fixed/improved, maybe also post updates new here. And only then maybe, after some time close it._

MatthiasKerbl, 2022-04-13:
Hi @KAMIKAZEUA,

good to hear you like the new update. :)

ad 1: The readout is precise, however, AlDente rounds it to whole percentages which should be accurate enough. 

ad 2: You are right this is a bug. We will fix it asap. 

The reason why I closed the issue after my first reply was because your first point about Sailing Mode was answered and we already had an issue/feature request regarding the Hardware Battery Percentage open. However, since you told me about a bug that is not yet fixed, I will open it again.  :)

KAMIKAZEUA, 2022-04-13:
Well, I love perfectionism, when it charged to 79.52% instead of 80.0%, it wasn't a good feeling)

KAMIKAZEUA, 2022-04-14:
@MatthiasKerbl 

And another thing for to be more precise: on real 74.55% it still will "think" because of rounding that it's like 75%, so will not start charging, but it should start charging since it's already sailed from 80% -> 75%(again going lower to 74%).
So it will start charge on after 74.49% at first.

MatthiasKerbl, 2022-04-14:
Hi @KAMIKAZEUA,

Since even the Hardware Battery Percentage from the battery management system is only a good guess of the real SoC of the battery and usually fluctuates in the range of a couple of percent, we decided to round it to whole percentages.

Cheers,
Matthias

KAMIKAZEUA, 2022-04-14:
Hey @MatthiasKerbl 
You mean real real battery state? Like if were theoretically measured with an external instrument? Well, yes li-ion will be +- something, depends on a current load. But anyway, Wattagio shows me really stable values during charge/discharge.

So, anyway, with settings of charge to 80% and sailing to 75% it will actually be like this:

 - sailing ->
- 74.56% = 75%
- 74.43% = 74% -> started charging +
- 79.45% = 79%
- 79.56% = 80% - > stopped charging || 

I think this is wrong.. should be really close to 75% and 80%, for example: 74.98% = 75% and 79.96% = 80%, so maybe different rounding like +- 0.05% < or >

KAMIKAZEUA, 2022-04-15:
Another thing I noted.

When charging reached **79.50%** (data from Wattagio) and AlDente displayed **80%** and charging **stopped**.
After **3-5 minutes**, the charge became **79.45%** (data from Wattagio), and AlDente already displayed **79%**. 
Those, It comes out very abruptly, after ~ 5 minutes the maximum charge dropped by **1%** (what the user sees from AlDente).

MatthiasKerbl, 2022-04-15:
> Hey @MatthiasKerbl You mean real real battery state? Like if were theoretically measured with an external instrument? Well, yes li-ion will be +- something, depends on a current load. But anyway, Wattagio shows me really stable values during charge/discharge.
> 
> So, with settings of charge to 80% and sailing to 75% it will actually was like this for me today:
> 
> * sailing ->
> * 74.56% = 75%
> * 74.43% = 74% -> started charging +
> * 79.45% = 79%
> * 79.56% = 80% - > stopped charging ||
> 
> I think this is wrong.. should be really close to 75% and 80%, for example: 74.98% = 75% and 79.96% = 80%, so maybe different rounding like +- 0.05% < or >

Hi @KAMIKAZEUA,

Again, we use mathematical rounding. Hence the behavior of AlDente you encounter.
I do not see the benefit for the goal of AlDente by changing the rounding.

Cheers,
Matthias

MatthiasKerbl, 2022-04-15:
> Another thing I noted.
>
> When charging reached **79.50%** (data from Wattagio) and AlDente displayed **80%** and charging **stopped**. After **3-5 minutes**, the charge became **79.45%** (data from [...]

Hi @KAMIKAZEUA,

Again, in my opinion, AlDente is accurate enough as it is right now and we are focusing on different improvements right now.

Cheers,
Matthias

KAMIKAZEUA, 2022-04-15:
Hey @MatthiasKerbl 

Sure, that's is just my comments.  
I just thought it would be very easy 🙄 to add such an option(which will not round value of %): 

[image] [image]

MatthiasKerbl, 2022-04-17:
> Hey @MatthiasKerbl
> 
> Sure, that's is just my comments. I just thought it would be very easy 🙄 to add such an option(which will not round value of %):
> 
> [image] [image]

Maybe we will implement something like that in the future.

MatthiasKerbl, 2022-04-17:
Hi @KAMIKAZEUA,

We just release AlDente 1.16.1 which has a fix for the bug which displays the wrong state for the Hardware Battery Percentage feature.
https://github.com/davidwernhart/AlDente/releases/tag/1.16.1

Cheers,
Matthias

closed 2022-04-17 by MatthiasKerbl

KAMIKAZEUA, 2022-04-17:
Hey @MatthiasKerbl 

Thanks for the update. 

I found a very minor issue with text while update, some encoding like. 
 
[image]

KAMIKAZEUA, 2022-04-17:
By the way, thank for Ukraine support, where I'm from 🇺🇦(last part of my nickname actually stands for country code).

MatthiasKerbl, 2022-04-17:
> Hey @MatthiasKerbl
> 
> Thanks for the update.
> 
> I found a very minor issue with text while update, some encoding like.
> 
> [image]

Thanks for telling us, we will try to do something about that with the next update.

KAMIKAZEUA, 2022-04-22:
> Hi @KAMIKAZEUA,
>
> Again, in my opinion, AlDente is accurate enough as it is right now and we are focusing on different improvements right now.
>
> Cheers, Matthias

Hey @MatthiasKerbl 

This is still like a bug, because I set **80%** max charge and app only goes up to **79%**: 

[image]

Charging started on 74-75% and stopped on 79%, which is like 1% wrong, so should I set **81%** max to achieve 80% :) ?

KAMIKAZEUA, 2022-04-23:
@MatthiasKerbl yes, for all these days since my last post, I have never seen **80%** after charge **complete**, it's always **79%**, but I've set **Charge Limit** = **80%**.

Please fix this, then finally this bug will be complete.