AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #653: AlDente pro paid version 1.15.2 , None of the features working.

vishwasvogga, 2022-04-10:
None of the features working on Mac intel OS version 12.3.1. Attaching the Debug file and OS details. Please help.
[Archive 2.zip](https://github.com/davidwernhart/AlDente/files/8458980/Archive.2.zip)

vishwasvogga, 2022-04-10:
I have set the sailing mode on but MAC continues to charge to 100% even after restart of MAC/AL Dente.

MatthiasKerbl, 2022-04-10:
Hi @vishwasvogga,

Please answer the following questions so that I can better assess your situation.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

vishwasvogga, 2022-04-11:
Describe the bug
I have freshly installed aldente pro version, on my freshly update macbook air. but none of the functions, such as Sail mode, limit mode or any feature is NOT working right from the begining.

To Reproduce
Steps to reproduce the behavior:

Go to to al dente.
Click on enable sail mode.
Attach the charger.
- NO CHARGE LIMITATION, MAC CHARGES right upto 100%
Expected behavior
- IT was supposed to maintain the charge between 70-80% even if charger is connected

Screenshots
[image]
[image]

MacBook (please complete the following information):

  MacBook Air (13-inch, 2017)
 12.3.1 (21E258)
 PRO
1.15.2
sail mode - enabled
heat protection - enabled
calibration - disabled
rest of the functions - disabled

Additional context
NOTHING IS WORKING.

MatthiasKerbl, 2022-04-11:
Hi @vishwasvogga,

This might be an issue with the helper.
Could you please generate and share a debug file? https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

vishwasvogga, 2022-04-11:
[debug_file_aldente.zip](https://github.com/davidwernhart/AlDente/files/8462167/debug_file_aldente.zip)
Here is the compressed debug file.
@MatthiasKerbl

MatthiasKerbl, 2022-04-11:
Hi @vishwasvogga,

Unfortunately, your MacBook model is currently not supported for most of the features of AlDente Pro (Only the Charge Limiter through the Intel Mode works). You can find the supported models in the FAQs here on Github or on our website. https://apphousekitchen.com/faq/

We are currently working on an update to bring the Pro features to older MacBooks as well. However, I am not able to give you a release date yet.

Cheers,
Matthias

closed 2022-04-11 by MatthiasKerbl

MatthiasKerbl, 2022-04-11:
> Same issue with Intel mode.. Can i get a refund :) ? @MatthiasKerbl

Hi @vishwasvogga,

Yes, you can get a refund, please write our support or our payment provider([assist@paddle.com](mailto:assist@paddle.com)) with your purchasing details. You can find our support email address in your purchasing emails.

If you are still interested in using AlDente, I am sure we can get it to work on your device with Intel Mode. :)

Cheers,
Matthias

vishwasvogga, 2022-04-11:
@MatthiasKerbl  I tried with intel mode, but charging is not getting limited to desired limit, Continues to charge up to 100%.
Here is the debug file in intel mode.

[aldentepro-debug-2022-04-11-21:22:34.pasta.zip](https://github.com/davidwernhart/AlDente/files/8465713/aldentepro-debug-2022-04-11-21.22.34.pasta.zip)

[image]

MatthiasKerbl, 2022-04-12:
Hi @vishwasvogga,

It looks like that for some reason AlDente cannot communicate with the helper program in your case.

To fix this, you have to reinstall AlDente completely. Please follow this uninstallation guide: https://apphousekitchen.com/how-to-uninstall-aldente-pro/

Afterward, download and install AlDente again. Activate the Intel Mode and the Charge Limiter should work.

Cheers,
Matthias

vishwasvogga, 2022-04-13:
@MatthiasKerbl  I did some sort of manual update of helper and aldente. Now intel mode is working limiting the charge within 5% of set value. Hope you come up with sail mode for my mac version. keeping the paid version with hope. thank you.

MatthiasKerbl, 2022-04-13:
Hi @vishwasvogga,

I am glad to hear that AlDente works now with the Intel Mode.
Again, I can not give you a release date but we are working hard on releasing more of the Pro features for older MacBooks as well.

Cheers,
Matthias