AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #654: “Stop Charging When Powered Off” feature not working on Intel MacBook Pro (2019, 16”)

Freesail21, 2022-04-12:
**Describe the bug**
I recently purchased Lunar Pro, then upgraded my MacBook Pro (2019, 16”) to Monterey OS v.12.3.1 a few days later. However, even though I keep the “Stop Charging When Powered Off” feature on with my power adapter plugged in, and always in clamshell mode, whenever I wake up, the battery is always at 100% and I’m forced to discharge it back to my 55% limit.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Charge Settings
2. Click on “Stop Charging When Powered Off” 
3. Leave in clamshell mode with power adapter plugged in overnight
4. See error (100% fully charged upon booting up)

**Expected behavior**
I expected the charge to be fully stopped at the current charge level upon shutting down, and the battery drain should be minimal as normally expected when not charging. 

**Screenshots**
https://ibb.co/Yp7PML0

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro (16-inch, 2019)
 - macOS version: 12.3.1 (21E258)
 - AlDente Pro or AlDente Free?: Pro
 - AlDente Version? 1.15.2
 - List of all enabled/disabled settings in AlDente: Enabled: Sailing Mode, Heat Protection, Stop Charging When Sleeping, Stop Charging When Powered off. Disabled: Disable Sleep Until Charge Limit, Intel Mode.

**Additional context**
Only started recently

andrewcrook, 2022-04-12:
@Freesail21 

fellow user here ... 

macOS 12.3.1 
M1 MBP 13” 2020
Ardent Pro 1.15.2

Not had any issues since macOS 12.3.1

But I notice you’re on Intel which has different power management hardware. 
I would try without `Intel Mode` a lot of functions don't work with it anymore and the charge limiter can be inaccurate and differ a lot because of the way it works (a different hardware path).

On my M1 Mac there is a period powering up and powering off where it can charge, this includes powering up and leaving it on the login screen. This can add up over days,  I leave it in sleep these days, but it doesn't sound like your issue?

Have you tested it by living it in sleep mode does it do the same? 
and do you get better results with Intel Mode disabled?

MatthiasKerbl, 2022-04-13:
Hi @Freesail21,

Thank you for the bug report. We have noticed some inconsistencies with the Stop Charging When Powered Off feature and are currently investigating it. I will let you know when we have new findings.

Cheers,
Matthias

MatthiasKerbl, 2022-10-04:
Hi @Freesail21,

We have fixed the bug you mentioned with the newest version of AlDente.

Cheers,
Matthias

closed 2022-10-04 by MatthiasKerbl