AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #658: More information about Battery

KAMIKAZEUA, 2022-04-13:
**Is your feature request related to a problem? Please describe.**
I'm looking for more technical information about battery: 

- Design Capacity - xxxx mAh
- Full Charge Capacity - xxxx mAh
- Manufacture date
- Age
- Cycle Count
- Power adapter(Connected) - XX Watts
- any other useful info

For Menu Bar icon:
- Show temperature
- Show wattage
- Show voltage
- Show amperage

**Describe the solution you'd like**
Since this app is all about Battery of my MacBook, so I think it should provide me more technical info about it.
For example: 

[image] [image][image] 

All this could be added into **contextual menu** on the app: 
[image]

And into the **Menu Bar icon**:

[image]

**Describe alternatives you've considered**
coconutBattery, Wattagio

andrewcrook, 2022-04-18:
you missed one showing battery level as estimated time

andrewcrook, 2022-04-18:
same as #250

KAMIKAZEUA, 2022-04-24:
@MatthiasKerbl perhaps should be useful https://github.com/sicreative/BatteryStatusShow

andrewcrook, 2022-04-24:
@KAMIKAZEUA

You have to be a bit careful you cannot just go around copying stuff. 
Open source is still licensed  and this is freemium with comcerial option.

MatthiasKerbl, 2022-04-24:
> @MatthiasKerbl perhaps should be useful https://github.com/sicreative/BatteryStatusShow

Hi @KAMIKAZEUA,

Thank you, will check it out.

MatthiasKerbl, 2022-04-24:
> @KAMIKAZEUA
> 
> You have to be a bit careful you cannot just go around copying stuff. Open source is still licensed and this is freemium with comcerial option.

Hi @andrewcrook,

Just wanted to say that we take open-source licenses seriously and would not just copy code from other projects without permission.

Cheers,
Matthias

andrewcrook, 2022-04-26:
@MatthiasKerbl 

Of course and I am sorry if it sounded otherwise.
I was just commenting on the number of posts where people are wanting features from other applications and that it is not always that easy or possible to fulfil.

KAMIKAZEUA, 2022-04-27:
@andrewcrook that ioreg info is the same for all apps, so retrieving code will be the same too. I just pointed to that app, which already have this implemented, so should be easily taken as example.

MatthiasKerbl, 2022-04-27:
> @MatthiasKerbl
> 
> Of course and I am sorry if it sounded otherwise. I was just commenting on the number of posts where people are wanting features from other applications and that it is not always that easy or possible to fulfil.

Hi @andrewcrook,

All good and thank you for raising awareness for this since it is not clear to most people.

Cheers,
Matthias

MatthiasKerbl, 2024-02-09:
Hi everyone,

We have just released AlDente 1.25 which includes a new widget for battery data. The new data widget can be activated in the stats category of in the settings. Also, this feature will be updated in the future to include even more data. To get AlDente 1.25 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-02-09 by MatthiasKerbl

Volodymyr-13, 2024-02-09:
@MatthiasKerbl 

I just tried to update..

[image]

Volodymyr-13, 2024-02-09:
@MatthiasKerbl ok it works, I like it but not fully, since I wanted actually this(this is from first post of this issue): 

[image]

MatthiasKerbl, 2024-02-09:
> @MatthiasKerbl
> 
> I just tried to update..
> 
> https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/3391456/cf0ae2ff-3849-4eca-b196-d64ce3523292
> 
> Upd.: did manual download from site.

Hi @Volodymyr-13,

Thanks for letting us know. The update functionality is fixed now.

Best,
Matthias

MatthiasKerbl, 2024-02-09:
> @MatthiasKerbl ok it works, I like it but not fully, since I wanted actually this(this is from first post of this issue):
> 
> [image]

Hi @Volodymyr-13,

Thank you for your feedback, we will consider it in the ongoing development and improvement of AlDente.

Best,
Matthias

Volodymyr-13, 2024-02-09:
@MatthiasKerbl looking forward into this. 

Also, you could add maybe some nice progress bars, to be more catchy and finally fix this(https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/678) and so I could finally not to use Wattagio and just use this app alone)

[image][image]

Volodymyr-13, 2024-04-12:
> Hi @Volodymyr-13,
> 
> Thank you for your feedback, we will consider it in the ongoing development and improvement of AlDente.
> 
> Best, Matthias

Any updates on this? 
I wish to add a System Load **W** same as Wattagio doing: 

[image]
[image]

Volodymyr-13, 2025-01-03:
@MatthiasKerbl can you please Open this issue? 

Since my last message bellow is still what I want from AlDente, this is really easy task, since you already have implemented this information in popover, so just add this text into Menubar is easy.