AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #659: Hardware Battery Percentage Toggle Doesn't Switch From "Activated" to "Deactivated" or vice versa

Macusercom, 2022-04-13:
**Describe the bug**
Toggling "Hardware battery percentage" doesn't change the string "deactivated" while toggling "sleep mode" above toggles the strings for both (sleep mode and hardware battery percentage).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to AlDente's settings
2. Click on "Charging"
3. Toggle "Hardware battery percent"

**Expected behavior**
Toggling either of these toggles should change the string "activated" to "deactivated" or vice versa only for the one setting it applies to.

**Screenshots**
https://abload.de/img/bildschirmfoto2022-04ncjap.png

**MacBook (please complete the following information):**
 - MacBook model: MacBookPro18,3 (2021 14" M1 Pro MBP)
 - macOS version: macOS Monterey 12.3.1 (21E258)
 - AlDente Pro
 - AlDente Version 1.16
 - List of all enabled/disabled settings in AlDente: Sailing Mode & Hardware Battery Percentage

MatthiasKerbl, 2022-04-14:
Hi @Macusercom,

Thank you for reporting this issue. We are already aware of it and will fix it soon.

Cheers,
Matthias

MatthiasKerbl, 2022-04-17:
Hi @Macusercom,

We just release AlDente 1.16.1 which has a fix for the bug you mentioned.
https://github.com/davidwernhart/AlDente/releases/tag/1.16.1

Cheers,
Matthias

closed 2022-04-17 by MatthiasKerbl