AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #661: Battery stays at 100% for hours after being unplugged.

est4tic, 2022-04-20:
Battery stays at 100% for hours after being unplugged.

[image]

MacBook M1 Pro
12.3.1
AlDente Free
1.16.1
Everything on default

MatthiasKerbl, 2022-04-21:
Hi @est4tic,

Might be a miscalibrated battery.
Take a look at our FAQs:
"HELP MY BATTERY CAPACITY DROPPED AND MY MACBOOK TURNS OFF WITH 50% LEFT OR STAYS AT 100% FOR HOURS!!!
Keeping your battery at a lower percentage, such as under 80%, over weeks without doing full cycles (100%-0%-100%) can result in a disturbed battery calibration. When this happens, your Macbook might turn off with 40-50% left or it will stay at 100% for hours and your battery capacity can drop significantly. However, this is only due to a disturbed battery calibration and not because of a faulty or degraded battery. To avoid this issue, we recommend doing at least one full cycle (0%-100%) every two weeks. Even if your battery calibration gets disturbed, doing 4-5 full cycles will recalibrate your battery and the capacity will go up again. AlDente Pro has a feature called Calibration Mode(https://apphousekitchen.com/feature-explanation-calibration-mode-2/), which will automatically do a full cycle when started. You can read more about the calibration of "smart" batteries here: https://batteryuniversity.com/article/bu-603-how-to-calibrate-a-smart-battery"

Cheers,
Matthias

est4tic, 2022-04-21:
Hi,

Thanks for the reply. I thought about this, but the MacBook is brand new. It finally went down... maybe it is not calibrated or maybe the battery is just way, way better then I expected it to be. I will do a calibration soon.

MatthiasKerbl, 2022-04-21:
Hi @est4tic,

I think it might be a combination of the battery being better than you expected and how the macOS battery percentage works. MacOS does not show you the hardware battery percentage, especially when you are at 100%. It will show you 100% even when the hardware battery percentage is down to 95%. AlDente Pro has a feature to show you the hardware battery percentage. You can check it out here: https://apphousekitchen.com/feature-explanation-hardware-battery-percentage/

I will close this issue for now. If you experience any issues again feel free to open it again.

Cheers,
Matthias

closed 2022-04-21 by MatthiasKerbl