AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #662: Screen brightness keeps shifting using Discharge

tomwolsky, 2022-04-21:
When I use Discharge in AlDente the screen brightness intermittently gets darker and brighter. I have Display System Preferences  set to automatic brightness. The room brightness isn't changing. The slider in Display moves up and down. The screen dims and brightens. If I shut down AlDente the brightness is constant. 

I expected the screen brightness to remain constant. 

Screen recording is too large. 70 MB. 

2021 M1 Max 14" MacBook Pro
macOS 12.4b2
AlDente Pro
[image]

tomwolsky, 2022-04-21:
I'm not seeing this in 12.3.1 on the same machine.

closed 2022-10-31 by MatthiasKerbl

city17, 2023-04-05:
I'm having the same issue that screen brightness dims intermittently since using AlDente Pro. It also always happens when I switch from a user account without AlDente turned on to my main user account which does have AlDente running. 

This is on a Macbook Pro 14 M1 2021. 

I have automatic screen brightness adjustment turned off on both accounts.

Any remedies/suggestions @MatthiasKerbl ?

MatthiasKerbl, 2023-04-08:
> I'm having the same issue that screen brightness dims intermittently since using AlDente Pro. It also always happens when I switch from a user account without AlDente turned on to my main user [...]

Hi @city17,

Do you have the macOS setting "Slightly dim the display on battery on" turned on?
If yes, try to turn it off.
Furthermore, if you want to have AlDente working on multiple user accounts, check out our tutorial **[here](https://apphousekitchen.com/fast-user-switching-explanation-and-tutorial/)**.

Cheers,
Matthias

city17, 2023-04-08:
> Hi @city17,
>
> Do you have the macOS setting "Slightly dim the display on battery on" turned on? If yes, try to turn it off. Furthermore, if you want to have AlDente working on multiple user accounts, [...]

Thanks Matthias, turning that setting off seems to have solved the problem as far as user-switching goes. Will install AlDente on the other user to and hopefully it works all normal then! Thanks!