AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #663: Does not know to scooch out from under the notch

madler, 2022-04-23:
AlDente on the menu bar will happily hide under the notch on an M1 MacBook Pro (14"). See photo:

[image]

I had to take a photo to show that, since interestingly a screen shot (command-option-4) does not try to simulate the notch. It shows AlDente on the command bar, even though there are no screen pixels there!

[image]

Steps to reproduce the behavior:
1. Buy an M1 MacBook Pro with a notch in the screen.
2. Install AlDente.
3. Have enough menu bar icons to the right of it for AlDente to be under the notch.

I would expect that the menu bar item would scooch to the left to be visible, like other ones. Not hidden.

MacBook Pro 18,4
macOS 12.3.1
AlDente Free 1.16.1
charge limit 80%

MatthiasKerbl, 2022-05-11:
Hi @madler,

Unfortunately, we are not able to reproduce this issue on our 16" M1 MacBook Pro. For us, the icon appears left of the last icon and right of the notch.

Do you still encounter this issue?
Did you try to reinstall AlDente?

Cheers,
Matthias

MatthiasKerbl, 2022-11-05:
Hi @madler,

Do you still experience the AlDente icon in the middle of the menu bar with our newest version AlDente 1.19.2?

Cheers,
Matthias

madler, 2022-11-06:
It seems to behave differently now. Instead of hiding under the notch, the menu bar icon just goes away. It does not scooch over to the other side of the notch. However googling seems to indicate that that is Apple's intended behavior.

What I am doing is turning on and off the Accessibility menu item in System Preferences. When it is off, I see AlDente. When it is on, the AlDente menu bar item is replaced by the Accessibility menu bar ico.

[image]

[image]

Nothing ends up getting hidden under the notch.

I am running macOS 12.6.1 and AlDente Free 1.19.2.

closed 2022-11-06 by madler

ysomad, 2023-10-20:
Getting the same issue with aldente pro 1.22.3
m1 pro 16

on open aldente appears and then dissapears behind the notch

[image]
[image]

venthura 13.5.1