AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #664: System stutters trying to maintain AlDente SoC

juloquinto, 2022-04-25:
**Describe the bug**
System stutters upon reaching state of charge set in AlDente

**To Reproduce**
Steps to reproduce the behavior:
1. Set SoC level in AlDente
2. Wait until SoC is reached and experience stuttering

**Expected behavior**
No stutter

**Screenshots**
Youtube and DaVinci Resolve stutter with playback, as well as audio sources

**MacBook (please complete the following information):**
 - A2485 16" M1 Max 10-core/32-gpu/64GB-ram
 - macOS 12.3.1
 - AlDente Free
 - AlDente Version 1.16.1
 - List of all enabled/disabled settings in AlDente: All disabled except Launch at Login, Show on Startup, and Show Dock Icon

**Additional context**
Losing focus on app with audio or video source (sent to background or while multitasking) will result in stuttering with playback after a few seconds while AlDente is trying to maintain desired SoC. I wish AlDente Free had a 7-day trial to see if this issue would go away with the Sailing Mode in the Pro version. Was about to buy a Pro/Lifetime License but thought to myself, this shouldn't be happening even in the Free version. It's the most basic function of the app, a paid Sailing Mode shouldn't be the fix.

MatthiasKerbl, 2022-10-31:
Hi @juloquinto,

Do you still experience this issue with our newest version AlDente 1.19.1?
Could you share a debug file when it happens? https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Cheers,
Matthias

MatthiasKerbl, 2023-01-01:
Hi @juloquinto,

I will close this issue for now since I haven't heard back from you in a while. I will open the issue again if you have any new findings.

Cheers,
Matthias

closed 2023-01-01 by MatthiasKerbl