AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #665: Sailing mode does not work

dalisoft, 2022-04-27:
**Describe the bug**
As title says, **Sailing mode** does not work nor indication does not work

**To Reproduce**
Steps to reproduce the behavior:
1. Download ALDente (`brew install aldente --cask`)
2. Register Pro account
3. Enable Sailing mode and configure your desired parameter
4. Connect external display or charger
5. Try work at least 4+ hours

**Expected behavior**
Battery value should be dropped but it does not and expecting see **Sailing mode** icon

**Screenshots**
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook model: MacBookPro17,1 (M1, 2020)
 - macOS version: 12.3.1 (21E258)
 - AlDente Pro or AlDente Free:  **AlDente Pro**
 - AlDente Version: v1.16.1
 - List of all enabled/disabled settings in AlDente: See screenshots

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2022-04-27:
Hi @dalisoft,

The Sailing Mode only starts working when you charge to the charge limit.
From your screenshots, I can see that you set the charge limit lower than the current SoC. This way, the Sailing Mode does not start.
Please try to set a Charge Limit which is higher than your current SoC and let your MacBook charge to the Charge Limit. When the Charge Limit is reached, Sailing Mode should start.

Cheers,
Matthias

dalisoft, 2022-04-27:
Hi @MatthiasKerbl 
SoC means System-on-Chip or what it does mean?

I triggered Down too and re plugged, does not work too

MatthiasKerbl, 2022-04-28:
> Hi @MatthiasKerbl SoC means System-on-Chip or what it does mean?
> 
> I triggered Discharge too and re plugged, does not work too

Hi @dalisoft,

I am sorry, i should have explained that SoC in battery terminology means State of Charge which is your current charge level or battery percentage.

Please try the following:

1. Unplug your MacBook and discharge your battery to 70%
2. Set your Charge Limit to 80%
3. Plug in your MacBook
4. When 80% is reached, do you see the little Sailing Mode icon like that?
[image]

Cheers,
Matthias

dalisoft, 2022-04-28:
@MatthiasKerbl I found tricky variant to get that icon and somehow worked in a short period **Sailing Mode**

1. Setting all desired configurations (see original post screenshots)
2. Click to "Reinstall" helper application
3. Reboot MacBook

And i can see icon and starts little but after that stops, like drops from **65%** to **63%** but not to **55%** as excepted

[image]

MatthiasKerbl, 2022-05-11:
> @MatthiasKerbl I found tricky variant to get that icon and somehow worked in a short period **Sailing Mode**
>
> 1. Setting all desired configurations (see original post screenshots)
> 2. Click to [...]

Hi @dalisoft,

From what I can see, Sailing Mode works as expected. Since you have an M1 MacBook, it will probably never completely go down to 55%.

Cheers,
Matthias

closed 2022-05-11 by dalisoft

drozo45, 2023-01-27:
@MatthiasKerbl I am experiencing exactly the same issue, but in sailing mode I also only see a drop by 2-3% (sometimes)
My configuration is charge limit 85%, sailing mode difference 35% (so it should go from 85 -> 50% before charging again). I only get down to 81-82% sometimes and then it says constant. Working on my laptop all day, so this is observed over 8+ hours. Using 14" MacBook M1 Pro w/ external display

Also tried to follow the instructions to let the laptop charge up to the charge limit first, then I see the sailing icon, but behaviour remains the same - most of the time it does not even discharge by 1%. 

Could you also explain what you meant when you said "Since you have an M1 Macbook, it will probably never go down to 55%" to @dalisoft

MatthiasKerbl, 2023-01-27:
> @MatthiasKerbl I am experiencing exactly the same issue, but in sailing mode I also only see a drop by 2-3% (sometimes) My configuration is charge limit 85%, sailing mode difference 35% (so it [...]

Hi @drozo45,

From your description, it sounds like the Sailing Mode is working as intended.
Please read the following article about the Sailing Mode on our blog here: https://apphousekitchen.com/feature-explanation-sailing-mode/
If you still have any questions, let me know.

Larzans, 2024-01-02:
@MatthiasKerbl Hi, i have the same situation as @drozo45 , i set the sailing mode to have a 30% difference with 70% being the max charge value.
It goes down to 60% and then basically stays there (exactly there according to iStat Menu).

So THIS is the expected behaviour, even though i set it to move between 70% and 40% ?
Quite misleading in my opinion.

That is not what i understood it to be at all, and it was honestly the only reason that i bought AlDente because i had similar problems with the 'battery' app.

MatthiasKerbl, 2024-01-02:
Hi @Larzans,

Did you read the [feature explanation](https://apphousekitchen.com/feature-explanation-sailing-mode/) of the Sailing Mode on our blog?

Here is a part of it that should explain your potential misunderstanding:

> Help, Sailing Mode doesn’t work on my MacBook!?
> The Sailing Mode is not intended to discharge your battery actively. Therefore, your MacBook will barely lose any energy even in Sailing Mode, since the power brick is the main source of energy. So do not worry when the battery percentage does not change even if the Sailing Mode is activated. Sailing Mode is still doing its thing.
> 
> Why doesn’t Sailing Mode keep the MacBook’s battery level fluctuating within a specific range?
> Contrary to popular belief, cycling a battery within a ‘safe’ range, like 30% to 70%, is actually more detrimental to its lifespan than maintaining it at a stable, optimal level, such as 80%. The cycling process accumulates charge cycles more rapidly, accelerating the wear and tear on the battery compared to leaving the MacBook connected at a stable and healthy charge. That said, this isn’t to suggest that you should refrain from using your MacBook on the go. While MacBooks are designed for mobility, it’s advantageous to keep it at a stable and healthy charge when using it in a stationary setting, like an office, rather than allowing it to continuously cycle.

> That is not what i understood it to be at all, and it was honestly the only reason that i bought AlDente because i had similar problems with the 'battery' app.

How did you understand the Sailing Mode to work and where did you read about it?
We are always trying to be as transparent and honest about the features of AlDente as possible to avoid any misunderstandings and not be misleading.

Best,
Matthias

Larzans, 2024-01-02:
Hi @MatthiasKerbl , thanks for the quick reply, i heard about it in a forum for battery (the os app).
When i then had a look at the blog post you linked and the Example 1 and what i took away from this sentence there:
"Sailing Mode is activated and an interval of 5% is set. The MacBook will not charge again until the lower limit of the interval, 75%, is reached." is that if i put an interval higher than 5%, for example 30%, the 'not charge again' part would then happen at 45%.
That is also what is suggested in the settings when it says: "Will sail to 45%", which it does not with the settings described there.

I understand now that the sailing mode is basically the same as the 'Charger Limit', with the difference that there are micro wait times between the charges, periodically disabling the charging, but basically also staying at a given percentage without discharging to the lower limit as it is suggested in the settings.

You might want to clarify that, this interval setting is very misleading and the wording of "Will sail to 45%" is just incorrect.

Unfortunately this was the only reason i bougth AlDente, if you are now telling me that it does not even do the discharging to 10% less thing then it is basically useless to me...

MatthiasKerbl, 2024-01-02:
Hi @Larzans,

"Sailing Mode is activated and an interval of 5% is set. The MacBook will not charge again until the lower limit of the interval, 75%, is reached."
This sentence is 100% correct, AlDente will prevent any energy going from the power adapter to the battery while the battery percentage is within the sailing range. However, this does not mean the energy for the MacBook is coming from the battery. This is not the case, instead, the power adapter will provide the energy for the MacBook. This behavior is the same for any chosen sailing range, for example, 30%.

> That is also what is suggested in the settings when it says: "Will sail to 45%", which it does not with the settings described there.

We try to be careful and precise with the words we choose in the settings. We are saying "Will sail to " and not "Will discharge to " indicating that the battery won't discharge due to using the MacBook.

> I understand now that the sailing mode is basically the same as the 'Charger Limit', with the difference that there are micro wait times between the charges, periodically disabling the charging, but [...]

This is not completely correct. There are no micro wait times between charges and also no periodically disabling the charging. Instead, while the MacBook is within the Sailing Range, AlDente will prevent any energy from the charger going to the battery. The reasons why the battery will still lose charge over time are self-discharge, the power usage of the MacBook is higher than the power provided by the charger, and the MacBook being unplugged.

> You might want to clarify that, this interval setting is very misleading and the wording of "Will sail to 45%" is just incorrect.

We came up with the phrase " sail to " in the context of batteries to not use discharge which is commonly known and means something else. That's why we also try to explain it as much as possible on our blog.

Why would you want a feature that discharges and charges the battery within a certain range?

Best,
Matthias

Larzans, 2024-01-03:
Hi @MatthiasKerbl, thanks again for indulging my doubts.

Sooo, just to sum it up in my own words:

When inside the 'sailing range' the sailing mode cuts power to the battery completely and the battery will not be charged until the lower limit is reached.
During sailing mode the charger supplies the laptop with the energy it needs and only in cases where it can't supply enough power for the laptop's needs the battery will be used to close the gap.
During sailing mode the battery is disconnected from the power supply and also from the laptop / consumers with the exception of the laptop's power demand being too high for the charger.

In sailing mode the battery is only loosing its charge due to self-discharge and if the power demand is higher than what the power the charger can supply.

So, theoretically i should see a discharge of the battery to 45% in the above scenario, albeit only after quite a long time (on modern laptops with moderate use).

Did i understand this correctly?

And also, are there any studies into the 'letting the battery be' vs. 'charging and discharging the battery within a certain range' that you can point me to?

Thanks again!