AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #667: AlDente crash followed by several toggles not functioning after relaunching

fiveangle, 2022-04-28:
AlDente set to either regular or Intel mode with 70% target charge level stops showing "70%" and shows "0%" within an hour after starting. Left running many days (generally 1-5 days), it crashes with following log...

<details><summary>[Crashlog Summary (full crashlog attached)]</summary>

Process:               AlDente [22005]
Path:                  /Applications/AlDente.app/Contents/MacOS/AlDente
Identifier:            com.apphousekitchen.aldente-pro
Version:               1.16.1 (26)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           AlDente [22005]
User ID:               504

Date/Time:             2022-04-28 02:42:34.295 -0700
OS Version:            macOS 11.6.5 (20G527)
Report Version:        12
Anonymous UUID:        AFFE86A5-86EC-526B-638D-922D056ABE00

Sleep/Wake UUID:       475BBD4D-5BE4-4FF6-8469-E1A3C853E85A

Time Awake Since Boot: 220000 seconds
Time Since Wake:       120000 seconds

System Integrity Protection: enabled

Crashed Thread:        4  Dispatch queue: com.apple.libdispatch-manager

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [22005]

Application Specific Information:
Kernel bug: The kernel cannot allocate memory needed for the pseudo-receive.

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff204e1ba2 __semwait_signal + 10
1   libsystem_c.dylib             	0x00007fff20461c1a nanosleep + 196
2   libsystem_c.dylib             	0x00007fff20461a82 sleep + 41
3   libxpc.dylib                  	0x00007fff2026f649 _waiting4memory + 14
4   libxpc.dylib                  	0x00007fff2025733f _xpc_mach_port_allocate + 92
5   libxpc.dylib                  	0x00007fff2025a59d xpc_connection_send_message_with_reply + 127
6   com.apple.Foundation          	0x00007fff213583ad -[NSXPCConnection _sendInvocation:orArguments:count:methodSignature:selector:withProxy:] + 3114
7   com.apple.Foundation          	0x00007fff2135f198 -[NSXPCConnection _sendSelector:withProxy:arg1:arg2:] + 135
8   com.apple.Foundation          	0x00007fff2135f10a _NSXPCDistantObjectSimpleMessageSend2 + 33
9   com.apphousekitchen.aldente-pro	0x000000010485a4af 0x10483b000 + 128175
10  com.apphousekitchen.aldente-pro	0x0000000104860b25 0x10483b000 + 154405
11  com.apphousekitchen.aldente-pro	0x0000000104884105 0x10483b000 + 299269
12  com.apphousekitchen.aldente-pro	0x0000000104882914 0x10483b000 + 293140
13  com.apple.Foundation          	0x00007fff213dce1f __NSFireTimer + 67
14  com.apple.CoreFoundation      	0x00007fff20624ca9 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
15  com.apple.CoreFoundation      	0x00007fff2062479d __CFRunLoopDoTimer + 927
16  com.apple.CoreFoundation      	0x00007fff206242fa __CFRunLoopDoTimers + 307
17  com.apple.CoreFoundation      	0x00007fff2060aed3 __CFRunLoopRun + 1988
18  com.apple.CoreFoundation      	0x00007fff2060a04c CFRunLoopRunSpecific + 563
19  com.apple.HIToolbox           	0x00007fff28852a83 RunCurrentEventLoopInMode + 292
20  com.apple.HIToolbox           	0x00007fff288527e5 ReceiveNextEventCommon + 587
21  com.apple.HIToolbox           	0x00007fff28852583 _BlockUntilNextEventMatchingListInModeWithFilter + 70
22  com.apple.AppKit              	0x00007fff22e12d72 _DPSNextEvent + 864
23  com.apple.AppKit              	0x00007fff22e11545 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
24  com.apple.AppKit              	0x00007fff22e03869 -[NSApplication run] + 586
25  com.apple.AppKit              	0x00007fff22dd7a6c NSApplicationMain + 816
26  com.apphousekitchen.aldente-pro	0x0000000104842c9c 0x10483b000 + 31900
27  libdyld.dylib                 	0x00007fff2052ff3d start + 1

Thread 1:: com.apple.NSURLConnectionLoader
0   libsystem_kernel.dylib        	0x00007fff204df2aa mach_msg_trap + 10
[... 122 lines cut ...]
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Graphics: kHW_NVidiaGeForceGTX750MItem, NVIDIA GeForce GT 750M, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.9)
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

</details>

Upon relaunch of AlDente, it correctly shows 70% again for target charge state, but several UI settings are no longer able to be used:

[image]

All UI elements on other pages are able to be used.

**Other details**
Calibration mode is never able to be finished once the battery state indicator spontaneously and incorrectly goes from "70%" to "0%", instead resulting in computer charging all the way to 100% and never entering discharge phase of calibration.

[AlDenteCrashlog.txt](https://github.com/davidwernhart/AlDente/files/8584649/AlDenteCrashlog.txt)

MatthiasKerbl, 2022-05-11:
Hi @fiveangle,

Which MacBook model do you have?

When Intel Mode is activated, all features except for the Charge Limiter, are not available anymore.
You can read more about it here: https://apphousekitchen.com/feature-explanation-intel-mode/

Cheers,
Matthias

MatthiasKerbl, 2022-10-31:
Hi @fiveangle,

Do you still experience this issue with our newest version AlDente 1.19.1?

Cheers,
Matthias

fiveangle, 2022-11-04:
Sorry for the delay. I have not used that MBP since filing. I will test with 1.19.1 and report back.

MatthiasKerbl, 2022-12-24:
Hi @fiveangle,

Since you have not reported back in a while, I will close this issue for now. However, feel free to get back and I will open the issue again.

Cheers, Matthias

closed 2022-12-24 by MatthiasKerbl