AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #671: On Discharge Mode, if I close laptop's screen, the monitor with usb-c won't wake up

Te0SX, 2022-05-01:
**Describe the bug**
I have one LG external monitor with usb-c which gives power to the MBP 14". I use AlDente Pro, paid.

Normally if I close my laptop's screen while I use my mac, the monitor usually also turns black/ready to turn off, until I press a key to the keyboard or move the mouse which wakes the monitor again. 

But, before a bit while I was having the Discharge Mode activated, I closed the laptop's screen and neither the keyboard or mouse could turn on the monitor. I thought what may be the trouble and since I had the issue with the stuck grayed Sleep (for which I had to use the command from [here](https://forums.macrumors.com/threads/sleep-greyed-out-mac-stopped-going-to-sleep.2195256/?post=28662800#post-28662800) ) I always suspect AlDente for everything :D . And it was, when I disable the Discharge Mode, everything is back to normal, I can wake my monitor.

**To Reproduce**
Steps to reproduce the behavior:
1.  Have your Macbook screen open and connect it with a monitor powered with USB-C 
2. Activate Discharge Mode
3. Close laptop's screen
4. Can't wake the monitor

**Expected behavior**
I should be able to wake up the monitor by pressing a key or moving/clicking the mouse

**MacBook (please complete the following information):**
 - MacBook 14" M1 Pro
 - macOS 12.3.1 (21E258)
 - AlDente Pro
 - AlDente  1.16.1
 - Default choices, only enabled Launch at Login and changed Appearance choices.

**Additional context**
Minor bug and would appear rarely to someone, it could be just for me anyway. Thought it wouldn't hurt to share it here.

poori, 2022-08-15:
Same problem for me. I have a macbook air m2, but slightly different variation.

Steps to reproduce this behavior:
1. m2 air plugged in to power, and also connected to usb-c monitor. battery should be close to target charge level.
2. go to sleep. 
3. wake up next day and laptop monitor will wake up, but external display connected via usb-c will not. 
4. unplug and replug usb-c cable to activate monitor again. 

Expected behavior
monitor should wake up along with laptop screen

Macbook
- Macbook M2 Air 
- 12.5
- aldente free
- 1.17.1
- 80% charge limit

Additonal context
probably likely to happen to anyone using a usbc cable to connect to the monitor. I think al dente is shutting off the usb-c along w/ the power cable but needs to re-enable the usbc monitor when it is waking up from sleep.

MatthiasKerbl, 2022-10-21:
> **Describe the bug** I have one LG external monitor with usb-c which gives power to the MBP 14". I use AlDente Pro, paid.
>
> Normally if I close my laptop's screen while I use my mac, the monitor [...]

Hi @Te0SX,

Thanks for sharing this detailed bug report.
We are aware of this bug and were so far not able to find a workaround or fix it. However, we will keep trying. In the meantime, I hope it is not too annoying and an easy possibility to avoid this bug is to activate Discharge after the lid was closed.

Cheers,
Matthias

MatthiasKerbl, 2022-11-03:
Hi @Te0SX,

We have just released AlDente Pro 1.19.2 which has a fix for the bug you mentioned.
Let me know if it works for you as well.

Cheers,
Matthias

Te0SX, 2022-11-06:
Hey Mattias,

Yeap, now it seems to be working fine, thanks. 👍 

Congrats for the fix and I wish you the best with the development.

Cheers,
Theodoros

closed 2022-11-06 by Te0SX

javierchan, 2024-09-25:
Hello, i have the same issue as @Te0SX but AlDente Pro 1.28.4, in a macbook M1 connected to an XDR Display (USB C) in macOS 15.0 (24A335), any workaround?