AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #672: "disable sleep until charge limit" discharges MBP completely

davidgirard, 2022-05-03:
Hello there,

It's all in the title. Happened to me yesterday, was working in bed, closed my 2019 MBP w/ ~50% left on it, woke up to a 0% battery. It would appear Al Dente was preventing the laptop to go to sleep so it could charge, except it wasn't connected to a charger. 

Version 1.16.1 (full/paid).

Cheers,
David

MatthiasKerbl, 2022-05-11:
Hi @davidgirard,

Please send us a debug file the next time it happens.
https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Cheers,
Matthias

MatthiasKerbl, 2022-09-01:
Hi @davidgirard,

Are you still experiencing this bug with our newest version AlDente 1.18?

Cheers,
Matthias

davidgirard, 2022-09-01:
Hi Matthias,

Thanks for the followup. 

I disabled the "Disable Sleep until Charge Limit" option as it prevented me from putting the MBP to sleep manually ("Sleep" option in the menu was blocked). It effectively provided a workaround for the bug.

I believe the bug was that my battery was <80% (my charge limit) and "Disable Sleep until Charge Limit" was on. 

Note that even with 1.18, if I enable "Disable Sleep until Charge Limit", the sleep option will sometime be available, and sometime it will be blocked. It seems random and I can't pinpoint exactly why. So I keep "Disable Sleep until Charge Limit" disabled.

Cheers,
David

MatthiasKerbl, 2022-09-01:
> Hi Matthias,
>
> Thanks for the followup.
>
> I disabled the "Disable Sleep until Charge Limit" option as it prevented me from putting the MBP to sleep manually ("Sleep" option in the menu was blocked). [...]

Hi @davidgirard,

Have you seen our blog article about this feature? https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/
It describes in detail when AlDente will disable sleep.

Cheers,
Matthias

davidgirard, 2022-09-01:
Matthias - I have. Somehow, the software isn't always properly sensing whether the laptop is on battery or plugged in.

Cheers,
David

MatthiasKerbl, 2022-09-01:
> Matthias - I have. Somehow, the software isn't always properly sensing whether the laptop is on battery or plugged in.
> 
> Cheers, David

That's odd, we have tested this feature thoroughly for many months now on different devices. Could you send us a debug file when it happens again?

Cheers,
Matthias

MatthiasKerbl, 2022-11-11:
Hi @davidgirard,

Any update on this?
Are you still experiencing this issue on our newest version AlDente 1.19.2?

Cheers,
Matthias

MatthiasKerbl, 2023-01-01:
Hi @davidgirard,

Since I haven't heard back from you in a while, I will close this issue for now. If you have any new findings, I will open it again.

Cheers,
Matthias

closed 2023-01-01 by MatthiasKerbl