AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #673: Process name of AlDente missing

marchpane, 2022-05-04:
**Describe the bug**
When I go in the aktivity manager and look for AlDente I can not see the name of it. Only the logo is visible.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the activity monitor
2. Search for AlDente

**Expected behavior**
Not only the logo is visible, but also the process name

**Screenshots**

[image]

**MacBook:**
 - M1 Air
 - Monterey 12.3.1
 - AlDente Pro 1.16.1

MatthiasKerbl, 2022-05-11:
Hi @marchpane,

Thanks for sharing, we are currently investigating this issue.

Cheers,
Matthias

MatthiasKerbl, 2022-05-13:
Hi @marchpane,

We could fix this issue and have just released a new version AlDente 1.17.1.
https://github.com/davidwernhart/AlDente/releases/tag/1.17.1

Cheers,
Matthias

closed 2022-05-13 by MatthiasKerbl