AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #675: Improve Heat Protection: Set temperature range

alexbaramilis, 2022-05-09:
Currently, if you set the Max. Temp to say 35 degrees, it will charge until it reaches 35 and then pause. But then it will immediately drop to 34.9 and start charging again for a few seconds until it reaches 35 again. So it will charge for a few seconds, pause for a few seconds, charge for a few seconds, etc, which I can't imagine it's good for the battery and doesn't really help if you're trying to charge.

One solution for this would be to set a temperature range for the heat protection. For example, you set the Max. Temp to 35 and you have another field for "Start charging when temperature drops again to" which you set to 32 for example. So it will charge until it reaches 35 and then it will cool down until it reaches 32 before it starts charging again. This way you give it a bit more of a breathing room to charge.

Another great alternative would also be to turn on the fans when it reaches Max. Temp as described in #630.

I love this app so many thanks and many 👍 for the good work!

hubreceived, 2022-05-11:
I like this suggestion

DamonHD, 2022-06-24:
Good idea.  Basically it should have a 'sailing' hysteresis...

Rgds

Damon

devklepacki, 2023-08-02:
This could be a customizable numeric value (or with just on/off), but potentially just a hardcoded feature for a faster implementation. Either way, I see two points here:
1. **The main question is, in fact, whether sailing when overheating would actually cause the battery to charge quicker, and is the current charging "logic" causing any harm.** Waiting for a cool down to a margin limit could actually take more time than just working just around the limit itself, as it is implemented currently.
2. I could see in some cases that this could cause you to actually never charge the battery if the limit is too wide. What I mean is that it may be easier for the temperature to vary between 34.9 and 35.2 than between 32 and 35.2°. So the battery could still charge in the first case, albeit slowly, but never charge if the battery wasn't able to cool down to 32 (someone's sailing margin).

MatthiasKerbl, 2024-02-09:
Hi everyone,

We have just released AlDente 1.25 which includes an update for the Heat Protection feature. The new updated Heat Protection feature has a built-in hysteresis to avoid switching between charging and charging paused all the time. Read more about the updated Heat Protection feature on our blog: [Feature Explanation - Heat Protection](https://apphousekitchen.com/feature-explanation-heat-protection/). To get AlDente 1.25 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-02-09 by MatthiasKerbl