AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #676: Ability to schedule Calibration Mode / Ability to have multiple saved settings profiles

hubreceived, 2022-05-11:
1. If possible, to let us schedule the calibration mode at specific time cycles (eg. Sat 3PM every week, and if the laptop isn't on, start calibration mode as soon as it's active -and if possible to add the history of the last 5 calibrations done and their dates and times-), because I'd rather not be bothered about remembering the last time I did one.

2. The ability to save at least one settings profile so I can quickly change my settings back to what I usually prefer, because sometimes changing some settings affects others.

Thanks a lot, the program is great.

ryaeng, 2022-05-28:
I too am interested in scheduled calibrations. With that feature, this app would be rock solid and I would purchase today.

I am willing to give some of my cycles to help with development. Let me know how I can help.

Bachsau, 2022-06-06:
Be aware that calibration also puts strain on your battery for very little benefit. There is no reason to do it, unless you notice the readings are absolutely flawed and it is a problem for you. As long as it doesn't become a problem you can't ignore, you don't have to remember to do it.

holografique, 2022-07-14:
> Be aware that calibration also puts strain on your battery for very little benefit. It is not something you should do on a regular basis. There is no reason to do it, unless you notice the readings are absolutely flawed and it is a problem for you. As long as it doesn't become a problem you can't ignore, you don't have to remember to do it.

This is contradictory to the advice that Aldente provides in their description of the calibration feature. Specifically stating that they recommend you cycling the battery through calibration mode at least once every two weeks if you make little to no use of battery powered operation, in order to keep the charge functionality healthy. Are you saying this is not the case?

That said, I also vote for being able to schedule calibration mode as my Macbook Pro is 98% used in clamshell mode as a desktop, with the occasional 2% use as a laptop on the go under battery power. So I too would love to not have to remember when was the last time I calibrated the battery.

Bachsau, 2022-07-14:
> they recommend you cycling the battery through calibration mode at least once every two weeks if you make little to no use of battery powered operation, in order to keep the charge functionality healthy.

No, they don't. It is said that it "_can_ prevent battery miscalibration from happening", which means to keep the readings accurate. That's not the same as "to keep the charge functionality healthy". 😉

You can always restore calibration by doing a full cycle, even if it's really messed up. Doesn't hurt the battery.

holografique, 2022-07-14:
> No, they don't. It is said that it "_can_ prevent battery miscalibration from happening", which means to keep the readings accurate. That's not the same as "to keep the charge functionality healthy". 😉

You may want to read through their current feature description and recommendations here: 
https://apphousekitchen.com/feature-explanation-calibration-mode-2/

Specifically...

> **When Should I Use The Calibration Mode?**
> If you barely use your MacBook battery-powered or use it only for a short time battery-powered, you should definitely use the Calibration Mode regularly to keep your battery calibrated.
> 
> **How Often Should I Use The Calibration Mode?**
> This primarily depends on how often you use your MacBook battery powered. If you do not use your MacBook battery-powered at all, you should use the Calibration Mode at least every other week.

That's a pretty clear implication that they are suggesting this be done for longevity of the battery. If you're suggesting otherwise, then might be good for AlDente to update their information. Cause right now, your suggestions and their suggestions don't align.

Bachsau, 2022-07-14:
Well, I'm not a battery expert. I just think you don't have to worry that much about it. One thing however, is clear: Doing full battery cycles wears them down over time, and it doesn't matter for what reason you do them.

MatthiasKerbl, 2023-03-12:
Hi everyone,

We have just released AlDente 1.21.1 which has a new beta feature called Schedule. This feature allows users to set specific Actions in AlDente to run automatically on predetermined days, times, and intervals. For example, Calibration Mode can be set to run every two weeks.
You can read more about the Schedule feature and its current limitations here: https://apphousekitchen.com/feature-explanation-schedule/
You can find a full list of the Release Notes here: https://apphousekitchen.com/changelog/

Cheers,
Matthias

MatthiasKerbl, 2026-10-08:
Hi everyone,

Scheduling the Calibration Mode is now possible with the Schedule feature, including weekly, every two weeks, monthly, and yearly intervals. AlDente also shows the date of the last successful calibration within the Calibration Mode and the Dashboard.

Since the main part of this request is now implemented, I will close this issue. If you would still like to see settings profiles, feel free to open a new feature request for it so that we can track it separately.

Best,
Matthias

closed 2026-10-08 by MatthiasKerbl