AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #677: Start of Beta Phase for AlDente Pro support for non Thunderbolt 3 MacBook Models

MatthiasKerbl, 2022-05-11:
Hi Everyone,

Over the last couple of months we worked hard on adding support of AlDente Pro to older Intel MacBooks as well and I am happy to announce that we were successful. Starting with AlDente Pro 1.17 we are starting the beta phase for supporting non Thunderbolt 3 Intel MacBooks. If you find any bugs, please let us know.

Take a look at the following article for more information about this new version of AlDente Pro: https://apphousekitchen.com/adding-aldente-pro-support-to-16-new-intel-macbook-models-full-list-below/

Full List Of Newly Supported MacBook Models:

**MacBook Pro**
MacBook Pro (Retina, 15-inch, Mid 2015)
MacBook Pro (Retina, 13-inch, Early 2015)
MacBook Pro (Retina, 15-inch, Mid 2014)
MacBook Pro (Retina, 13-inch, Mid 2014)
MacBook Pro (Retina, 15-inch, Late 2013)
MacBook Pro (Retina, 13-inch, Late 2013)

**MacBook Air**
MacBook Air (13-inch, 2017)
MacBook Air (13-inch, Early 2015)
MacBook Air (11-inch, Early 2015)
MacBook Air (13-inch, Early 2014)
MacBook Air (11-inch, Early 2014)
MacBook Air (13-inch, Mid 2013)
MacBook Air (11-inch, Mid 2013)

**12″ MacBook**
MacBook (Retina, 12-inch, 2017)
MacBook (Retina, 12-inch, Early 2016)
MacBook (Retina, 12-inch, Early 2015)

Cheers,
David and Matthias

tosbsas, 2022-05-12:
that means Catalina MacBooks - is that right? Right now I am on free 2.2 - can I test the pro 1.17 to see if it works?

MatthiasKerbl, 2022-05-12:
> that means Catalina MacBooks - is that right? Right now I am on free 2.2 - can I test the pro 1.17 to see if it works?

Hi @tosbsas,

macOS 11 Big Sur or higher is still required.

Cheers,
Matthias

tosbsas, 2022-05-12:
uyyy - too bad

closed 2022-08-10 by MatthiasKerbl