AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #679: Ver. 1.17 Pro doesn't launch when offline

CBDAN, 2022-05-12:
**Describe the bug**
AlDente 1.17 Pro doesn't launch when the Mac is offline, because the app needs to establish a connection with paddleapi.com.

DamonHD, 2022-05-12:
This also seem likely to be what blocks logging in for 30s or more, since I usually will not have a working WiFi session running when I open up my MBA.  It is intensely annoying since one of the nice features of macOS has traditionally been that it's very quick to resume.  Please fix: defer licence checking until later.

MatthiasKerbl, 2022-05-13:
Hi @CBDAN and @DamonHD,

We could fix this issue and have just released a new version AlDente 1.17.1.
https://github.com/davidwernhart/AlDente/releases/tag/1.17.1

Cheers,
Matthias

closed 2022-05-13 by MatthiasKerbl

DamonHD, 2022-05-13:
Thank you.  I have installed 1.17.1 and I'll see how it works for me!

Rgds

Damon

DamonHD, 2022-05-17:
Doesn't seem to have made a difference for me.  Last few MBA openings at least 8s delay to flashing password prompt.  Without AlDente running, no delay.

Rgds

Damon

MatthiasKerbl, 2022-05-17:
> Doesn't seem to have made a difference for me. Last few MBA openings at least 8s delay to flashing password prompt. Without AlDente running, no delay.
> 
> Rgds
> 
> Damon

Hi @DamonHD,

your issue does not seem to be correlated to this issue.
Could you open a new GitHub issue with a precise description of what exactly happens in which scenario?

Thanks,
Matthias