AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #680: Cannot install the helper successfully

fragrusti, 2022-05-13:
I use AlDente since a while on my M1 Macbook Air. I purchased the Pro Lifetime license since it is a very useful piece of software.

It was working perfectly, suddendly it returned to its original icon (the spaghetti one) and lost all the information and the helper.

Then I tried to reinstall the helper: even if it says that it is installed correctly, nothing happens and the battery bar remains grey.

Moreover i tried to remove and reinstall the helper and aldente itself - nothing. I tired version 1.17 and 1.17.1 nothing changes.

I attach the content of my pasta (pure genious!) debug file (renamed in txt otherwise GH would not let me to attach it) and a screen shot of the greyed battery bar. [aldentepro-debug-2022-05-13-12:44:16.pasta.txt](https://github.com/davidwernhart/AlDente/files/8686454/aldentepro-debug-2022-05-13-12.44.16.pasta.txt)

[image]

Thank you in advance,
FrAg

MatthiasKerbl, 2022-05-13:
Hi @fragrusti,

your debug file looks good and the helper communication seems to work fine.
As simple as it might sound, could you try the following:

1. unplug your MacBook from the power supply
2. turn off your MacBook
3. close the display 
4. wait for at least 2 minutes
5. restart your MacBook and plug it in again

Does AlDente work again?

Cheers,
Matthias

fragrusti, 2022-05-17:
Dear Matthias, 

it worked! Thank you so much. Very appreciated.

Ciao,
FrAg

closed 2022-05-17 by MatthiasKerbl

yogithesymbian, 2024-11-18:
how to permanent install ? when restart its always need to install again so sometime i forgot then charge until 100% damned

wangxiang4, 2025-09-22:
> how to permanent install ? when restart its always need to install again so sometime i forgot then charge until 100% damned mbp m1 sequoia

[Stuck at install helper app after macos update](https://github.com/AppHouseKitchen/AlDente-Battery_Care_and_Monitoring/issues/1030#issuecomment-3317143286)