AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #682: AlDente no longer respects max charge % and simply keeps on charging up

basmets6, 2022-05-16:
**Describe the bug**
Since yesterday, AlDente (Pro) no longer follows the limits/settings that I've set in the app. I have it set to charge max 70%.
However, now it simply continues past the 70%, with no signs of stopping.

It flashes between non-charging (power adapter) and then actively topping up. It switches constantly.
This didn't happen before at all, since now.

**To Reproduce**
... As mentioned above

**Expected behavior**
It should stop at the 70% charge limit that I've put

**Screenshots**
https://imgur.com/a/seaJQYi

**MacBook (please complete the following information):**
 - MacBook model: MBP 2020, 13" Intel
 - macOS version: 12.3.1
 - AlDente Pro
 - AlDente Version: 1.17.1
 - List of all enabled/disabled settings in AlDente: https://imgur.com/a/RgPgwnQ

**Additional context**

PeterAT00, 2022-05-16:
I have the same problem!

MatthiasKerbl, 2022-05-16:
Hi @basmets6,

Did you try to completely reinstall AlDente Pro?
Also, could you send us a debug file?

Cheers,
Matthias

MatthiasKerbl, 2022-05-17:
> Hi Matthias, thanks for the reply. I've reinstalled the helper a couple of times, reset the settings.. I'll go for a full reinstall now, I guess.
> 
> Here's the debug file: https://we.tl/t-Uov4icEYWD

Looking through your debug file, it looks like macOS really wants to charge your MacBook.

Do you have "Optimized battery charging" deactivated in your macOS Settings?

Did you do a macOS update recently?

basmets6, 2022-05-17:
Thanks for the reply. As for your comments – well, yes? Not sure what to reply to that in fairness..

I didn't touch any settings whatsoever since the day I purchased the app (AlDente Pro). This was ± 1 month ago.
It was working perfectly until just this Sunday, when I noticed this happening.

– I always had "Optimized battery charging" turned off.. That's why it worked in the beginning.
– Now I had a look at _Preferences > Battery > Battery > Battery Health_ and then turned off "Manage battery longevity". This was not disabled previously (but again, I didn't touch anything so it must've been enabled when AlDente was working fine).
– Didn't update anything. I've been traveling and honestly the WiFi was just awful (also roaming charges are hella expensive) so I could barely do that. I now see there's a new update 12.4 – I'll install that and see if it makes a diff.

Would be great to get a better understanding.. I also don't seem to be the only one with the issue (see other comment above). A bit useless to have purchased the Pro version when it simply doesn't work..

Holdsworth24, 2022-05-18:
Same situation here.

huyz, 2022-05-18:
Things may have started breaking since the macOS 12.4 update?

MatthiasKerbl, 2022-05-18:
> For me, I'm pretty sure it started breaking right after macOS 12.4 update, which was last night.

Hi @huyz,

Take a look at our FAQs:
"ALDENTE DOES NOT WORK ANYMORE AFTER MACOS UPDATE!
Unfortunately, for a small number of users, AlDente temporarily does not work after a macOS update. To get AlDente to work again, just reinstall AlDente completely and everything should work as expected again. If that did not help, try to shut down your MacBook completely and unplug the power brick for a couple of minutes. Afterward, restart your MacBook, plug it in, and everything should work again."

Cheers,
Matthias

MatthiasKerbl, 2022-05-18:
> Thanks for the reply. As for your comments – well, yes? Not sure what to reply to that in fairness..
>
> I didn't touch any settings whatsoever since the day I purchased the app (AlDente Pro). This was [...]

Hi @basmets6,

I am sorry if I did not explain it in detail but from what we could see, your debug file looks fine except for macOS trying to overwrite AlDente and keep your MacBook charging. This usually only happens, when "Optimized battery charging" is activated.
- "Manage battery longevity" only sometimes interferes with AlDente when you charge over 95%
- Usually, this behavior also happens after macOS updates but this should not be the case for you.

In general, offering the functionality AlDente has is more difficult than most people think, and even though we have implemented multiple mechanisms to avoid as many bugs/issues as possible, for some people issues still occur, and even though the outcome might look exactly the same, it does not mean at all that the cause of the problem is the same. Therefore, I need as much information as possible for every bug/issue from every user to be able to assess the situation.

Cheers,
Matthias

PeterAT00, 2022-05-18:
After completely removing and reinstalling the software it now works perfectly again. Also after upgrading to MacOS 12.4.

closed 2022-05-19 by MatthiasKerbl

huyz, 2022-05-19:
@MatthiasKerbl 

> Take a look at our FAQs: "ALDENTE DOES NOT WORK ANYMORE AFTER MACOS UPDATE! Unfortunately, for a small number of users, AlDente temporarily does not work after a macOS update. To get AlDente to work [...]

Thanks for that.

But note, a Github README page is not one of the first  places I'd look for an app's FAQ.  I would look here first: https://apphousekitchen.com/faq/ and that particular question isn't there.

MatthiasKerbl, 2022-05-19:
> 

Hi @huyz,

Thanks for bringing this to my attention, I will add this question on our website as well.

Cheers,
Matthias

MatthiasKerbl, 2022-05-19:
> @MatthiasKerbl
>
>
> Thanks for that.
>
> But note, a Github README page is not one of the first places I'd look for an app's FAQ. I would look here first: https://apphousekitchen.com/faq/ and that [...]

Hi @huyz,

Thanks for bringing this to my attention, I will add this question on our website as well.

Cheers,
Matthias