AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #686: AlDente not working - battery at 100%

pkowal-x, 2022-05-21:
**Describe the bug**
AlDente set up for different values, like 88 or 92%, but battery always charged to 100%.
This is a second time I file the bug, this time on another machine. This makes me wonder if the software works at all.
[image]
[image]

**MacBook (please complete the following information):**
M1 MacBook Air
MacOs 12.3.1
AlDente Free, but this way it will never be Pro

MatthiasKerbl, 2022-05-21:
Hi @pkowal-x,

I am sorry you are having trouble with AlDente.
There are multiple reasons why the Charge Limiter might not work for you.

Do you have Apple's "Optimized Battery Charging" disabled?
Did you do a macOS update recently?
Are you aware that the Charge Limiter in AlDente Free does not work when your MacBook is in sleep or powered off?
Did you try a complete reinstall of AlDente?
Did you try to turn off your MacBook, close the lid, unplug it, wait for 5 minutes, and turn it on again?
Could you send us a debug file to get more details about your issue?

Cheers,
Matthias

ovq, 2022-06-12:
I see this too, at least since 12.4. 
My MBP is set to 70% (sail 10%), but it always stays on 100% with AD running. I have restarted multiple times, now I just tried the 5' shutdown/unplug suggestion from above. Let's see.

enovikoff, 2022-06-13:
It is not clear to me that “launch at login” means that Al Dente doesn’t remain running after logout (or lock.)  Other apps solve the problem of remaining available while the MBP is on (with periods of sleep) so why can’t Al Dente do it?  The conflicting information in this thread also makes it unclear whether Al Dente works when the MBP is asleep, or if this behavior depends on whether one uses the free or paid version.  In my experience when I hit the Touch ID/lock button while my MacBook is attached to the charger or dock, it successfully prevents a full charge and the MBP seems to be sleeping. The most recent problem occurred when I used the MBP on battery and closed the lid when I was done, then plugged it in later and observed that it was charging towards 100% after I used Touch ID to unlock it and start using it, with “launch at login” off.   Turning that setting on might have prevented this problem, but I still am puzzled why it’s necessary, since other system utility apps just resume without issue.

MatthiasKerbl, 2022-10-04:
Hi @pkowal-x,

Do you still experience this issue with our newest version AlDente 1.18?
https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2022-10-16 by MatthiasKerbl

pkowal-x, 2022-10-16:
> Hi @pkowal-x,
> 
> Do you still experience this issue with our newest version AlDente 1.18? https://github.com/davidwernhart/AlDente/releases
> 
> Cheers, Matthias

I confirm, that the issue is resolved.

maps10, 2025-06-14:
> Hi [@pkowal-x](https://github.com/pkowal-x),
>
> I am sorry you are having trouble with AlDente. There are multiple reasons why the Charge Limiter might not work for you.
>
> Do you have Apple's [...]

I am on MacBook Air m2 and I tried the free version of the ardent and it doesn't work