AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #687: Update 1.17.1 lose settings for Charge limit

KAMIKAZEUA, 2022-05-23:
**Describe the bug**
Update 1.17.1 lose settings for Charge limit
It was on charging state at around 76% when I fired update process. After app relaunch I saw it's not charging, so I right clicked on menu bar and it started to charge, but I noticed also that it's now 100% limit, checked it with settings, gave me that screen-shoot. 

**To Reproduce**
Steps to reproduce the behavior:
1. Update to 1.17.1
2. Open Settings of the app
3. Notice now it's 100%
4. Set it to 80%, notice it's still showing 95% sail.

**Expected behavior**
Settings not lose after update.

**Screenshots**

Screenshoot taken after I realized that settings are lost, so I set from 100% to 80% limit.

[image]

**MacBook (please complete the following information):**
 - MacBook Pro M1 2021
 - macOS version 12.4
 - AlDente Pro 
 - AlDente 1.17.1

jmvdkolk, 2022-05-31:
@KAMIKAZEUA. I have similar issues and need to open a support case. I think the problem is an incompatibility between AlDente and MacOS version 12.4. I did not have issues when still on MacOS version 12.3. Problems started with MacOS version 12.4.

MatthiasKerbl, 2022-10-31:
Hi @Volodymyr-13,

It seems like you changed the Charge Limit to 100% after the update with the right-click on the menu bar icon.
Do you know about the Right-Click feature of AlDente Pro?
[image]

Did you experience this issue with the latest updates?

Cheers,
Matthias

MatthiasKerbl, 2022-12-24:
Hi @Volodymyr-13,

Did you experience this issue throughout the last couple of updates or do you think that it might be possible that you triggered the Right-Click feature unintentionally?

Cheers,
Matthias

jmvdkolk, 2022-12-25:
Hi Matthias,

Thank you for following up. I am very happy to say that I have not experienced this issue throughout the last couple of updates. Thank you for the amazing support.

Cheers,
Jean-Marc

closed 2023-01-01 by MatthiasKerbl