AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #688: Calibration never getting to 100%

xXValiXx, 2022-05-25:
**Describe the bug**
Calibration not working. If I start the calibration it charges the battery to 100% on the macOS battery status icon. On the AlDente Icon it says 99%. After waiting 30 minutes it still says 99% on the AlDente. The calibration is not continuing. It stays at "Charge to 100%" task.<br>
**To Reproduce**
1. Starting MBP
2. Plugin your charingbrick
3. Open AlDente and click "Start Calibration"
4. See problem

**Expected behavior**
AlDente should charge up to 100% and then start to discharge to 15% and so on.

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBookPro18,4
 - 12.3.1
 - AlDente Pro 
 - 1.17.1
 - Enable: Sailing Mode, Heat Protection, Stop charging when sleeping, Stop charging when app closed, Hardware Battery Percentage, Charge Limit

**Additional context**
Not sure but when I stop "Calibration"(Battery 100% or 99%) and my "Charge Limit" is set to 75% it does not automatically discharge. I need to click discharge in AlDente to make this happen. Maybe not an expected behaviour.

Lderib, 2022-06-24:
I'm experiencing the same issue but first step is stuck forever at 96%.

Lderib, 2022-06-26:
Hey, fyi, I found a workaround by disabling Hardware Battery Percentage.

MatthiasKerbl, 2022-06-28:
> **Describe the bug** Calibration not working. If I start the calibration it charges the battery to 100% on the macOS battery status icon. On the AlDente Icon it says 99%. After waiting 30 minutes it still says 99% on the AlDente. The calibration is not continuing. It stays at "Charge to 100%" task. **To Reproduce**
> 
> 1. Starting MBP
> 2. Plugin your charingbrick
> 3. Open AlDente and click "Start Calibration"
> 4. See problem
> 
> **Expected behavior** AlDente should charge up to 100% and then start to discharge to 15% and so on.
> 
> **Screenshots** [image]
> 
> **MacBook (please complete the following information):**
> 
> * MacBookPro18,4
> * 12.3.1
> * AlDente Pro
> * 1.17.1
> * Enable: Sailing Mode, Heat Protection, Stop charging when sleeping, Stop charging when app closed, Hardware Battery Percentage, Charge Limit
> 
> **Additional context** Not sure but when I stop "Calibration"(Battery 100% or 99%) and my "Charge Limit" is set to 75% it does not automatically discharge. I need to click discharge in AlDente to make this happen. Maybe not an expected behaviour.

Hi @xXValiXx,

Which MacBook model do you have?
Did you only wait 30 minutes? Sometimes it can take 1h+ to go from 99%-100%.

Cheers,
Matthias

MatthiasKerbl, 2022-06-28:
> I'm experiencing the same issue but first step is stuck forever at 96%.

Hi @Lderib,

Which MacBook model do you have?
Do you have the settings "Manage Battery Longevity" activated in macOS settings?

Cheers,
Matthias

xXValiXx, 2022-06-29:
> Hi @xXValiXx,
>
> Which MacBook model do you have? Did you only wait 30 minutes? Sometimes it can take 1h+ to go from 99%-100%.
>
> Cheers, Matthias

Hey @MatthiasKerbl 

I´ve got the MacBook Pro 14" (2021) with M1 Max, 64GB RAM and 1TB SSD, 96W Power Adapter
I didn´t try it over 1h for sure. But i did wait more then 30 minutes.
But i can try to wait longer then 1 hour.

Cheers,
Fabian

xXValiXx, 2022-06-30:
Hey @MatthiasKerbl ,
I got the calibration done. It just took over 1 hour to charge from 99 to 100%. I am wondering why this is takes so much time.
It took me more than half a day to complete the calibration because the macbook battery wont go down.
I openend some apps and games to drain faster. Hope this wont be an issue with the calibration.
Thanks for your reply. 

Cheers,
Fabian

Lderib, 2022-06-30:
> > I'm experiencing the same issue but first step is stuck forever at 96%.
> 
> Hi @Lderib,
> 
> Which MacBook model do you have? Do you have the settings "Manage Battery Longevity" activated in macOS settings?
> 
> Cheers, Matthias

Hello @MatthiasKerbl ,

I have a MacBook Pro (16-inch, 2021) Apple M1 Max.
The "Optimised battery charging" option is unchecked. Think this is what you're referring as "Manage Battery Longevity"

regards,
Luc

Lderib, 2022-06-30:
> Hey @MatthiasKerbl , I got the calibration done. It just took over 1 hour to charge from 99 to 100%. I am wondering why this is takes so much time. It took me more than half a day to complete the [...]

Hello @xXValiXx ,

Do you have the "Hardware Battery Percentage" option enabled in Al Dente?
Regards,
Luc

xXValiXx, 2022-07-06:
Hello @Lderib,

yes the Hardware Battery Percentage option is enabled.
Cheers,
Fabian

MatthiasKerbl, 2022-08-20:
Hi @xXValiXx,

We released AlDente Pro 1.18 a couple of days ago which has an update for the calibration while Hardware Battery Percentage is enabled.
https://github.com/davidwernhart/AlDente/releases/tag/1.18

Cheers,
Matthias

closed 2022-08-20 by MatthiasKerbl