AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #689: Haven't receive license email

wing-kai, 2022-05-30:
Hi, I made the purchase of AlDente Pro, and the credit card notification showed that the the deal is done. But I haven't receive any emails yet (I have check the junk mail box too), I also try to recent the email but it said "no order", I'm wondering what's wrong with my order?

credit card last four number: 6743
email: cyj9385@hotmail.com
Transaction time: UTC+8 2022-05-30 11:11

MatthiasKerbl, 2022-05-30:
Hi @wing-kai,

I am sorry you did not receive the license code.
I have sent the emails again and you should have already received them.

Cheers,
Matthias

closed 2022-05-30 by MatthiasKerbl

3lMghraby, 2022-05-30:
> Hi @wing-kai,
> 
> I am sorry you did not receive the license code. I have sent the emails again and you should have already received them.
> 
> Cheers, Matthias

could you look mine too
my email is 3lmghraby@gmail.com
same issue I can't activate my license and I paid and transaction is successful 
could you send the license to me again
thanks

MatthiasKerbl, 2022-05-31:
> > Hi @wing-kai,
> > I am sorry you did not receive the license code. I have sent the emails again and you should have already received them.
> > Cheers, Matthias
> 
> could you look mine too my email is [3lmghraby@gmail.com](mailto:3lmghraby@gmail.com) same issue I can't activate my license and I paid and transaction is successful could you send the license to me again thanks

Hi @3lMghraby,

Please take a look at my response here #690.

Cheers,
Matthias