AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #696: Dark / Light mode (config(

SergioSuarezDev, 2022-06-05:
**Describe the bug**
Can't see the widget corretly
¿It is possible select the dark or white mode by config?

**To Reproduce**
Steps to reproduce the behavior:
Enable dark mode only in menubar, not in rest of apps or system
Open the widget in menubar

**Expected behavior**
Good colors

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model:  MacBook Pro (15-inch, 2017)
 - AlDente Pro or AlDente Free? Free
 - AlDente Version? 1.17.1

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2022-09-01:
Hi @SergioSuarezDev,

For me, the colors still work flawlessly when I only use dark mode in the menu bar.

[image]

Is there any possibility that this is caused by something else?

Cheers,
Matthias

MatthiasKerbl, 2022-10-31:
Hi @SergioSuarezDev,

Do you have any updates on this?
Do you still experience this issue with our newest version Aldente 1.19.1?

Cheers,
Matthias

MatthiasKerbl, 2022-11-06:
Hi @SergioSuarezDev,

Since I did not hear back from you, I will close this issue for now but feel free to give any updates on this.

Cheers,
Matthias

closed 2022-11-06 by MatthiasKerbl