AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #699: AlDente 1.17.1 prevents MacBook Pro M1 2021 from charging until reboot

RichieB2B, 2022-06-09:
**Describe the bug**
My 16" MacBook M1 2021 running macOS 12.4 will not charge ever again after AlDente stops the charging at the set limit op 60%. After this the "plug -" symbol is shown. Even if the charger is disconnected the symbol still shows the "plug -" symbol. After the charge falls below the set sailing limit of 40% plugging in the charger does not start a charge. System Report shows:
```
AC Charger Information:

  Connected:	No
  Charging:	No
```

**To Reproduce**
Steps to reproduce the behavior:
1. Enable Sailing Mode 
2. Set Charge Limit to 60%
3. Plug in charger, wait until AlDente stops the charging at 60%
4. Unplug the charger, wait until the battery discharges until below the sailing limit
5. Plug in the charger
6. Check System Report and observe macOS does not detect the charger 

**Expected behavior**
Charging should start again after plugging in the charger when the battery is below the sailing limit.

**Screenshots**
n/a

**MacBook:**
 - MacBook model: MacBook Pro M1 2021
 - macOS version: 12.4
 - AlDente Pro
 - AlDente Version: 1.17.1
 - List of all enabled/disabled settings in AlDente:
 
[image]

**Additional context**
Stopping AlDente does not resolve the situation. Restarting the MacBook Pro enables the charging again.

jogu, 2022-06-10:
I've been having issues where my 2018 15" i9 MacBook Pro is refusing to charge (and also seemed to be cured by a reboot), but I'm on aldente pro 1.14.

I'm pretty sure the problems only started after upgrading to macOS 12.4.

RichieB2B, 2022-06-10:
For me as well: with macOS 12.3 this did not happen, I have only seen this on macOS 12.4.

bbm-design, 2022-06-12:
seen the same on 
  Model Identifier:	MacBookPro18,1
  Chip:	Apple M1 Pro
  Total Number of Cores:	10 (8 performance and 2 efficiency)
  Memory:	16 GB
  System Firmware Version:	7459.121.3
  OS Loader Version:	7459.121.3
  running Macos   12.4 (21F79)

rgdev, 2022-06-17:
Same here, 1.17.1 on MacOS 12.4 (M1 Pro MBP). Cannot charge anymore, this need to be fixed asap.
No issue before updating to 12.4 either.

Ariandr, 2022-06-27:
Guys, it's not related to AlDente. It's related to MacOS 12.4. I have this issue without AlDente installed.

@RichieB2B @rgdev @bbm-design

skynet01, 2022-06-28:
Yep you are right, here is the apple support forum of people complaining about this issue, pretty annoying https://discussions.apple.com/thread/253922483?login=true&page=3

RichieB2B, 2022-06-28:
Thanks @Ariandr it looks indeed like this is a generic macOS 12.4 issue. I only experienced it for a few days, did not come back for > 2 weeks now thankfully.

closed 2022-06-28 by RichieB2B