AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #701: MacBook can't sleep anymore

mvaneijgen, 2022-06-13:
**Describe the bug**
I have my laptop plugged in most of the time while using it, but if not using it and I know I'll need it later I just close it and unplug everything. Lately when I then come back to it and open it it will boot up from start and will notify mee that my laptop shut down and if I want to reopen all the programs.

I've just completely removed AlDente and tested it again and now this never happens anymore.  

**To Reproduce**
Steps to reproduce the behavior:
1. Set AlDente to charge until 75 ~ 80% 
2. Close laptop unplug everything 
3. Come back later and open laptop again (maybe four hours)

**Expected behavior**
The laptop function normally and boot from sleep 

Version 12.3 (21E230)
MacBook Pro (15-inch, 2017)
Processor 3,1 GHz Quad-Core Intel Core 17
Memory 16 GB 2133 MHz LPDDR3
Graphics Radeon Pro 560 4 GB
Intel HD Graphics 630 1536 MB
AlDente Pro 1.17.1 (Settapp)

RichieB2B, 2022-06-20:
I see the same thing with my MacBookPro18,2 running macOS 12.4 running AlDente Pro 1.17.1:

1. Sailing Mode is enabled
2. Stop Charging when Sleeping is disabled
3. Disable sleep until Charge Limit is disabled
4. Unplug power cable during sailing (below charge limit, above sail limit)
5. Close lid
6. Open lid a few days later
7. Battery is almost empty

ybhat, 2022-06-25:
Same issue for me. Sleep is greyed out. Al Dente Pro. Only after removing helper I was able to sleep. I use my MBP in Clamshell mode so it's annoying to see it completely discharged every time.

clintwood, 2022-07-21:
Same issue for me - had to remove AlDente to stop this happening!

redamancy-w, 2022-08-04:
I also have this situation, I changed the version to 1.13, and reinstalled, it got better

MatthiasKerbl, 2022-11-06:
Hi @mvaneijgen,

Do you still experience this issue with our newest version AlDente 1.19.2?
We have made a bunch of changes and implemented more safety mechanisms over the last couple of versions to prevent this from happening.

Cheers,
Matthias

mvaneijgen, 2022-11-08:
Just tested it and my PC seemed to stay in sleep all night while plugged in. Going to test while not plugged in and closed later.

MatthiasKerbl, 2022-11-09:
> Sadly same issue. My laptop was about 50% charged last night and when I open it in the morning it is still charged, but I get the message "You shut down your computer because of a problem. Do you want to reopen all the programs."

That's strange. Thanks for testing it.
When you open the MacBook again in the morning, which charge level/battery percentage did you have?

mvaneijgen, 2022-11-09:
I didn't pay attention to it. Its so automatic in the morning. I'll test again tomorrow.

mvaneijgen, 2022-11-10:
My laptop was at 55% when closed and when opened in the morning 45% was left. It agin restarted my computer in between that time.

MatthiasKerbl, 2022-11-11:
> My laptop was at 55% when closed and when opened in the morning 45% was left. It agin restarted my computer in between that time.

Thanks for the details. Unfortunately, I am still not able to reproduce this behavior. Could you generate and share a debug file when this happens the next time? https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Cheers,
Matthias

mvaneijgen, 2022-11-13:
Here you go. I had to add .log as a file extension, others wise I couldn't share it. 

[aldentepro-debug-2022-11-13-11:20:51.pasta.log](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/9996947/aldentepro-debug-2022-11-13-11.20.51.pasta.log)

MatthiasKerbl, 2022-11-14:
Hi @mvaneijgen,

Unfortunately, we could not find any abnormalities in your debug file. Most likely, because your system just restarted before you saved the debug file.
If possible, could you try to trigger this issue again but try to open your MacBook and generate a debug file after a while when it lost more battery than it should have but did not shut down yet?

Cheers,
Matthias

mvaneijgen, 2022-11-14:
I will try. I was also thinking, it only seems to happen after a really long while. If I just leave my laptop for a few hours nothing happens. Only when I leave it over night, this will happen, maybe the date flipping to a new day causes the issue? Just thinking out of the box.

MatthiasKerbl, 2022-11-14:
> I will try. I was also thinking, it only seems to happen after a really long while. If I just leave my laptop for a few hours nothing happens. Only when I leave it over night, this will happen, maybe the date flipping to a new day causes the issue? Just thinking out of the box.

Thanks. Theoretically possible but unlikely.

MatthiasKerbl, 2022-12-24:
Hi @mvaneijgen,

Do you have any new updates about this issue or have you tried my suggestion from above?

Cheers,
Matthias

mvaneijgen, 2022-12-24:
I went down a other rabbit hole, where I thought Better Touch tool was the culprit. https://community.folivora.ai/t/macbook-crashes-while-sleeping/29463/11 It seems to only happen when that app is on, but I'm still testing. 

[aldentepro-debug-2022-12-24-16:39:20.pasta.log](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/10298830/aldentepro-debug-2022-12-24-16.39.20.pasta.log)

MatthiasKerbl, 2022-12-26:
> I went down a other rabbit hole, where I thought Better Touch tool was the culprit. https://community.folivora.ai/t/macbook-crashes-while-sleeping/29463/11 It seems to only happen when that app is [...]

Interesting, keep me updated if you find out anything new.

Cheers,
Matthias

MatthiasKerbl, 2023-03-21:
Hi @mvaneijgen,

Do you have any updates about this issue? 

Cheers,
Matthias

mvaneijgen, 2023-03-21:
I do not. Upgraded to a new laptop M2 and now every works as expected

MatthiasKerbl, 2023-03-21:
> I do not. Upgraded to a new laptop M2 and now every works as expected

Sounds good.
Then I will close this issue.

Cheers,
Matthias

closed 2023-03-21 by MatthiasKerbl