AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #703: Random Hang and Restart

mctosima, 2022-06-14:
**Describe the bug**
I experience a random hang, followed by 100% fan speed (noticeable by the roar) and restart. This is very random. About once in a day. Normal activity (browsing, chatting, zoom call)

**To Reproduce**
Can't be reproduced

**MacBook (please complete the following information):**
 - MacBook model : Macbook Pro 14 inch with M1 Pro
 - macOS version 12.4
 - AlDente Pro or AlDente Free? AlDente Pro
 - AlDente Version? 1.17.1
 - List of all enabled/disabled settings in AlDente: Sailing Mode On (5%) , Heat Protection On (38C), Hardware battery percentage ON

**Additional context**
-

MatthiasKerbl, 2022-09-01:
> **Describe the bug** I experience a random hang, followed by 100% fan speed (noticeable by the roar) and restart. This is very random. About once in a day. Normal activity (browsing, chatting, zoom [...]

Hi @mctosima,

This sounds strange. Why do you think this is caused by AlDente?

Cheers,
Matthias

mctosima, 2022-11-01:
> Hi @mctosima,
> Any update on this?
> Do you still experience this behavior?
> Do you see any connection with AlDente?
> Cheers,
> Matthias

Hi Matthias

Since last month I'm no longer been using Aldente. Also, my battery maximum capacity now is 90%, while it hasn't been a year since I purchased this mac.

Compared to my last mac (mbp 2020), which has 95% battery max capacity in a year without Aldente.

I don't know whether this is related to Aldente or not, but I want to try using my mac without Aldente for some upcoming months.

MatthiasKerbl, 2022-11-01:
> Hi Matthias Since last month I'm no longer been using Aldente. Also, my battery maximum capacity now is 90%, while it hasn't been a year since I purchased this mac. Compared to my last mac (mbp [...]

Hi @mctosima,

Regarding battery degradation, I am seeing more and more misunderstandings and misconceptions about this topic. I have tried to answer some common questions in the thread here: #770 
I will also release some articles about this topic on our blog over the next couple of months.
No proof but an example is that our M1 Pro machine still has 98% capacity after a year even though we use it a lot to test new versions of AlDente.

I will close this issue for now but let me know if you have anything new to share about the original issue, then I will open it again.

Cheers,
Matthias

closed 2022-11-01 by MatthiasKerbl