AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #705: After completely uninstalling Aldent, my Mac cannot trigger optimized battery charging.

Loving-Vincent, 2022-06-23:
**Describe the bug**
After completely uninstalling Aldent, my Mac cannot trigger optimized battery charging. Does anyone have the same situation as me?

**To Reproduce**
Steps to reproduce the behavior:
1. uninstall AlDente refer to this [post](https://apphousekitchen.com/how-to-uninstall-aldente-pro/) for steps 
2. Open Mac Optimized Battery Charging Settings

**Expected behavior**
The Mac learns your daily charging pattern and holds charging above 80% until you need to use the battery.

But it never works.

**Screenshots**

[image]

**MacBook (please complete the following information):**
 - MacBook Pro (16-inch, 2019)
 - macOS Monterey
 - AlDente Free V1.16

MatthiasKerbl, 2022-06-23:
Hi @Loving-Vincent,

This has nothing to do with AlDente.
After uninstalling AlDente and activating Apple optimized battery charging, your MacBook basically tries to find a pattern in your usage and adapts charging to it. Therefore, if you would plug in your MacBook every day at 10 pm and unplug it every day at 8 am, macOS should charge your MacBook to 80% first and start charging it fully a few hours before 8 am. However, this is highly unreliable and gets disturbed easily when you do not have a daily charging pattern.

Cheers,
Matthias

Loving-Vincent, 2022-06-23:
Alright, i will keep watching, thanks

closed 2022-06-23 by Loving-Vincent

bitA23187, 2023-09-08:
Hi @Loving-Vincent, 
Have you solved this problem? I seem to have encountered the same problem after uninstalling AlDente, specifically, 

①I uninstalled it in an irregular way (delete AlDente (1.22) directly in Applications folder), and then triggered the problem; 
②then I reinstalled it and uninstalled it according to the [tutorial](https://apphousekitchen.com/how-to-uninstall-aldente-pro/), but the problem remained unresolved.
③ I then also tried reinstalling and uninstalling a newer version of AlDente (1.22.2), which failed either way.

Just to be clear, my mac is plugged in all the time, and usually the charging optimization is triggered within a week. Of course I'm not 100% sure that the issue is caused by AlDente, but I haven't found a similar cause of the problem so far.

**MacBook (please complete the following information):**

- MacBook Pro (14-inch, 2021)
- macOS Ventura 13.5.1
- AlDente Free V1.22/1.22.2

Loving-Vincent, 2023-09-12:
Not yet, i gave up :-(

bitA23187, 2023-09-12:
Thank you for your reply. At present, it seems that this problem can only be solved by reinstalling macos (#780 ). I will try it sometime.

Loving-Vincent, 2023-09-12:
This will be a tough job and probably I will never try it, but thank you.