AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #708: App does crazy things an my new M1 Pro (MacBook Pro 14)

ange335, 2022-06-26:
I upgraded from a MacBook Air M1 to a MacBook Pro 14 inch with M1 Pro (A2442) and the app is set up correctly. I used Apple Migrationassistent after updating the new Mac, as always. All is working perfect except AlDente Pro. I also deleted the app (fully, also in Applications Support in Libary and so on) and reinstalled it - nothing changed.

Heat protection is active and set to 33 degrees, battery temperature exceeds, Mac is still charging. Battery percentage is limited to 66%, Mac stops charging for a few minutes, than continiue charging, stops again, continiue again and so on until it reach his 100%… 

Currently macOS 12.5 Beta 4 is installed on booth devices, AlDente works like a charm on the Air but absolutly not on the Pro…

It looks and feels like the app wants to control the charging behavier but macOS takes over and sends the app into their place… 

Very unsatisfactory because the charging cycles went up dramatically when i want to avoid permanent 100% battery charge because can not use the limited charging function. Additional, the device is brand new a special configured and ordered model (RAM, SSD…)

What can i do to get it working asap? (Coding and deep macOS knowledge is available) Thx!

Regards from fu***ing hot Linz/Autria 😎
Andi

ange335, 2022-06-26:
Here you can see what I mean... @davidwernhart

Normally the Mac is not allowed to charge since the temperature is to high and the battery percentage is to high too. Pls see the following screenshot.

[image]

And at the pic below can you see what is happening when AlDente Pro sent the charger in standby... It  is not in standby instead of this it is charging with lowered charging power (69W power supply)

[image]

Ariandr, 2022-06-27:
Hi @ange335 
Turn off optimized battery charging from the Macbook preferences.

When i turned it off, nothing takes control over AlDente.

ange335, 2022-06-28:
This feature is turned off since I use AlDente... Of course it is off! 

Strange wise after 2-3 days of battery management doing what it wants to do finally it works - nothing changes, nothing touched. It simply works now as it does on the Mac Book Air M1.

I leave this open for now, may the developer has an idea what is happening and why this is happening. Maybe the bug is a feature, haha :)

Ariandr, 2022-06-28:
@ange335 
Well, probably macOS does extra work for some time on battery and then stops

ange335, 2022-06-28:
Maybe, but this have to be a bug from the Beta combined with the hardware from the MBP 14 inch M1 Pro because on the normal M1 in the Air it worked properly also under macOS 12.5 b4... Anyway - now it is working again as it should! Thx

closed 2022-06-28 by ange335