AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #709: Inaccurate charging limit

saenchon40, 2022-06-28:
My MacBook Air M1

Running macOS Monterey 12.4

I use AlDente set charge limit is 75% and Charge Settings Enable all options.

But the battery level is always 77% or 78%.

How can I fix this?

MatthiasKerbl, 2022-10-21:
Hi @saenchon40,

With AlDente Pro 1.19 we have introduced the feature Automatic Discharge, which will always automatically discharge your MacBook to the set Charge Limit. Therefore, the charge limit will be much more accurate.

Cheers,
Matthias

MatthiasKerbl, 2025-02-06:
Hi @saenchon40,

Do you still experience this issue with our latest version of AlDente?

Best,
Matthias

saenchon40, 2025-02-06:
Yes, I still experience this issue, but I resolved it by using the Automatic Discharge mode, and it works well.

saenchon40, 2025-02-06:
@MatthiasKerbl 
Yes, I still experience this issue, but I resolved it by using the Automatic Discharge mode, and it works well.