AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #711: Macbook forgot to have a battery included

juniorgary, 2022-06-29:
**Describe the bug**
I was using AlDente for some months and was happy with it. I decided to update to a newer version. The app asked for reinstalling the helper app which I did without hesitating. Afterwards the battery icon disappeared from the menubar. In the Systeminformation app all information about the battery (which is for sure still integrated) is empty or zero (capacity, loading cycles,...). If I disconnect the power plug the computer immediately shuts down as if there was no battery at all.
I'm quite frustrated and would appreciate your help! Do you have any idea how to repair this issue? Thank you in advance!
[image]

Maxung, 2022-07-02:
You might want to try [this](https://support.apple.com/en-us/HT201295), if you still need help

Danby108, 2022-07-05:
If resetting the SMC doesn't help, as Maxung basically suggested, then there could be a hardware disconnect between the battery and the laptop. In this case you could look up the replace the battery instructions on ifixit.com that are applicable to your particular MacBook Pro. Within those instructions they'll mention how to electrically connect, disconnect, or reconnect your laptop battery.

If that still doesn't work, even after reconnecting it and giving it time to charge, etc., you might find more things to try on that website. And you could also pick up a new replacement battery from them, if that seems to be the problem.

juniorgary, 2022-07-09:
Thank you, Maxung and Danby108 for your comments! Unfortunately resetting the SMC doesn't help. I replaced the original battery 1 year ago (according to instructions on ifixit.com), because it was swollen. To run the new one better, I just installed AlDente. This worked fine until I just installed the update. Since the problem occurred immediately after the update, I assume that it is a software problem. However, I will check the electrical connection.

MatthiasKerbl, 2022-11-11:
Hi @juniorgary,

I am sorry this happened to you.
As far as I remember, you are the first one to report a battery completely dying while using AlDente. Since we are only using technical pathways Apple implemented, I am having a hard time imagining that AlDente is the actual cause of this issue. Furthermore, when you completely uninstall AlDente and perform an SMC reset, all traces of AlDente are gone.
Do you have any updates on this?
Did you look into a potential connection issue between the battery and the MacBook?

Cheers,
Matthias

juniorgary, 2022-11-11:
Since I could not find a solution regarding the software, I ordered a new battery. After installation, the problem was gone. I don't know how intelligent the battery electronics are, but I can imagine that they classified the battery as defective because I had permanently limited the charge to 40% using Aldente. For this reason I no longer use Aldente (and hope that the battery will survive my Macbook, which is now 8 years old).

MatthiasKerbl, 2022-11-12:
Hi @juniorgary,

Thanks for the details.

> Since I could not find a solution regarding the software, I ordered a new battery. After installation, the problem was gone.

In my opinion, a lot of things point toward a defective battery. Especially, when a new one was immediately recognized again with no issues.

> I don't know how intelligent the battery electronics are, but I can imagine that they classified the battery as defective because I had permanently limited the charge to 40% using Aldente.

In my opinion, this is highly unlikely. To my knowledge, the battery management system in macOS does not classify a battery as defective, when its SoC was permanently limited. Furthermore, many people use AlDente to limit charging, however, you are the first and only one so far to report this issue occurring while using AlDente. Moreover, if macOS would classify a battery as defective, it would still detect it but would show "Service Recommended". You can read more about this here https://support.apple.com/en-us/HT204054. Therefore, due to your battery not being detected at all, this points towards some kind of hardware issue, either the battery itself or the connection between the battery and the MacBook. But again, this is just my opinion.

Hope I could clarify some things.

Cheers,
Matthias

MatthiasKerbl, 2022-12-24:
Hi @juniorgary,

Since everything seems to point toward a defective battery and not AlDente, I will close this issue. However, feel free to add any new findings.

Cheers,
Matthias

closed 2022-12-24 by MatthiasKerbl