AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #712: Charged to 100% after Calibration, even if limit was set

Volodymyr-13, 2022-07-01:
**Describe the bug**
Charged to 100% after Calibration, even if limit was set 

**To Reproduce**
Steps to reproduce the behavior:
1. Start-finish calibration
2. Notice 81% charge
3. Close the lid 
4. Open next day - notice 100% battery charge.

**Expected behavior**
Charge limit should work after calibration.

**Screenshots**

[image]

**MacBook (please complete the following information):**
 - MacBook Pro 14"M1
 - macOS 12.4
 - AlDente Pro ?
 - AlDente 1.17.1?
 - List of all enabled/disabled settings in AlDente:

[image]

jymr92, 2022-07-09:
Me, too.

andiweli, 2022-07-11:
Me too.
Pro License and keeps charging - limit is 80%. But charged up to 86 now even if I press "Discharge button".

Volodymyr-13, 2022-07-11:
Discharge actually worked for me, only this.. is ok..

andiweli, 2022-07-11:
Discharge starts discharging for a few seconds, then aborts and continues charging.
I am on 89% now - set the limit to 90% - perhaps it stops again.

andiweli, 2022-07-11:
Nope... charging happily ... 91% now.
Settings will be overruled.

schwedi, 2022-08-07:
Me too,
86% and there will always be short phases with the + icon , and it continues to charge
MacBook Pro 2017 15",  macOS Monterrey 12.5
[image]

iamkevindu, 2022-10-01:
Same issue here. Al Dente goes back to charging mode even after I click the discharge button.

MatthiasKerbl, 2022-11-02:
Hi everyone,

We have just released AlDente Pro 1.19.2 which has a fix for the Calibration Mode charging to 100% after the calibration bug.

Cheers,
Matthias

closed 2022-11-02 by MatthiasKerbl