AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #713: An option to allow automatic discharge

stepanjakl, 2022-07-01:
Hello,

First of all, thank you for an amazing app - I am super grateful! I got the Pro version for my new MacBook. (This is honestly something Apple should have as a part of their OS by default.)

So, I understand that the battery is managed only when the AlDente app is on. Once the MacBook is turned off, it fully recharges back. Fair enough. I am curious though if there was a way to change the firmware a.k.a. Apple BIOS to limit battery charge even when shut off, but plugged in?

Now, when the device is turned back on (with 99% or 100% battery charge), it doesn't automatically start **discharging** the battery. I wonder why? It would be really sweet to have an option to allow automatic discharge. I also wonder, if it is still "healthy" for the battery to discharge and charge every day from 80% to 99%?

I read through your blog where it is mentioned that the best way to use the app, hence manage the battery, is to put the laptop to sleep. Again, fair enough. However, I think it's pretty healthy for the device (as well as psychological well-being HAHA) to turn it off  after a day of work (may be good for cache clearing etc. too?). What do you think?

I am looking forward to the answer :)

DamonHD, 2022-07-01:
For your first point, have you seen the "Stop charging when sleeping" option?

Rgds

Damon

stepanjakl, 2022-07-01:
@DamonHD Yes, I have that option enabled actually. However, it only works when you put the laptop to **sleep**, not when it is **turned off**. My main request is to have an option to **enable discharge** by default when the app starts.

stepanjakl, 2022-08-03:
No response from the team?

mikebore, 2022-08-09:
_"Now, when the device is turned back on (with 99% or 100% battery charge), it doesn't automatically start discharging the battery. I wonder why? It would be really sweet to have an option to allow automatic discharge"._

Can I request this as well please.

MatthiasKerbl, 2022-10-21:
Hi @stepanjakl,

> First of all, thank you for an amazing app - I am super grateful!

Thank you for these kind words, we really appreciate it. :)

> So, I understand that the battery is managed only when the AlDente app is on. Once the MacBook is turned off, it fully recharges back. Fair enough. I am curious though if there was a way to change the firmware a.k.a. Apple BIOS to limit battery charge even when shut off, but plugged in?

In general, this is correct. However, the behavior during sleep and powered off is different for different MacBook models. Furthermore, we have implemented features to cope with and sometimes circumvent these limitations. You can read more about this topic in the following blog posts:
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/
https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/
https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
https://apphousekitchen.com/adding-aldente-pro-support-to-16-new-intel-macbook-models-full-list-below/

> Now, when the device is turned back on (with 99% or 100% battery charge), it doesn't automatically start discharging the battery. I wonder why? It would be really sweet to have an option to allow automatic discharge. 

With the newest update, we have just released a new feature called Automatic Discharge. Which will do exactly what you suggested if activated.

> I also wonder, if it is still "healthy" for the battery to discharge and charge every day from 80% to 99%?

In general, battery research indicates that charging to 100% on a regular basis and staying at 100% is rather bad for battery longevity. Therefore, I would avoid this charging behavior.

> I read through your blog where it is mentioned that the best way to use the app, hence manage the battery, is to put the laptop to sleep. Again, fair enough. However, I think it's pretty healthy for [...]

You are right, there are definitely also advantages when you turn off your MacBook over night. However, from a battery longevity perspective, when it means that during this time the battery will be charged to 100% and stay at 100% for a longer time on a regular basis then I would not recommend doing it or at least unplugging the MacBook when it is turned off.

Hope that helps and I hope it is okay that I close this issue now since we have implemented the mentioned feature.

Cheers,
Matthias

closed 2022-10-21 by MatthiasKerbl

stepanjakl, 2022-10-23:
Hello @MatthiasKerbl,

That's a really awesome update! Thank you for taking in all the feedback, answering the questions and above all implementing the additional discharge feature. You guys rock 🤘